davesteele/comitup #1: SSID selection goes to unencypted

hblanken, 2017-04-03:
Hi David, I am testing your comitup image.

When I logon to 10.42.0.1 I enter my SSID it takes me to the unencrypted page and I cannot enter a password. I manually removed the 'un' and it worked fine, but still a tiny bug to remove.

http://10.42.0.1/confirm?ssid=XXX&encrypted=unencrypted

davesteele, 2017-04-04:
Does that edit fix it?

tehDrew, 2017-06-20:
It did not fix it for me; glad I happened to glance at the url and see "unencrypted" and manually changed it.  Using WPA2 PSK.

davesteele, 2017-06-21:
As it happens, I've run across this as well, with a new Wifi Router 

To check if my fix works for you (183637d0), check if the command 'sudo iw dev wlan0 scan' shows an 'RSN:' section for your AP's SSID.

tehDrew, 2017-06-21:
it does show an RSN section under my router

davesteele, 2017-06-21:
Then it should be fixed in the next release.

tehDrew, 2017-06-21:
Thanks, davesteele!!  This is a very handy image you've made for those of us with wireless Pi's.  Your work is much appreciated :1st_place_medal:

davesteele, 2017-07-30:
fixed

closed 2017-07-30 by davesteele