davesteele/comitup #2: Total allocated sectors 15237120 greater than the maximum 3096576.

9jobsworth, 2017-07-12:
hello david,
it is me again.
my micro sd card is perfectly ok.
i ran fdisk on the image file which i downloaded and got this:-

fdisk  "2017-03-02-raspbian-jessie-comitup-0.7-03-13.img" 

Welcome to fdisk (util-linux 2.25.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): v
Total allocated sectors 15237120 greater than the maximum 3096576.

Command (m for help): p

Disk 2017-03-02-raspbian-jessie-comitup-0.7-03-13.img: 1.5 GiB, 1585446912 bytes, 3096576 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x432b3940

Device                                                                 Boot  Start      End  Sectors  Size Id Type
2017-03-02-raspbian-jessie-comitup-0.7-03-13.img1        8192   137215   129024   63M  c W95 FAT32 (LBA)
2017-03-02-raspbian-jessie-comitup-0.7-03-13.img2      137216 15237119 15099904  7.2G 83 Linux

Command (m for help): 

this would explain why i have been unable to dd your disk image to a micro sd card.

best wishes
john
ps still struggling to use ssh on a raspberry pi zero W.

davesteele, 2017-07-13:
Hmm. Indeed it is. The lite image also has a partition that exceeds the size of the device. My script's final partition resize is failing for some reason.

As it turns out, it doesn't matter, for two reasons. First, the filesystem on the partition does not go past the end of the device. Second, Raspbian has a first-boot behavior that resizes the partition and filesystem to fit the size of the SD card. Despite the dire warnings, it should work fine.

This issue does not explain why you can't dd the file:
  - Use dmesg to see what the last usb device is, to find the /dev/sd&lt;letter&gt; destination.
  - Make sure that you first umount any filesystems that may have automounted first.

I just burned and mounted that image. It worked fine.

davesteele, 2017-07-17:
In any case, there is a new image uploaded, which doesn't have this problem.

closed 2017-07-17 by davesteele

9jobsworth, 2017-07-17:
hello david,

i have discovered from the internet that the reason why i cannot use ssh 
running raspbian over wifi

is that it is necessary to add

IPQoS 0x00

at the ends of /etc/ssh/sshd.conf and /etc/ssh/ssh.conf.

and then it works.

i have not tried your new image file but i will.

i am just going out to buy another micro sd card.

i seem to consume a lot of them.

best wishes

john

davesteele, 2017-07-17:
I've never heard of the need to set QOS to get SSH working. Good catch.