davesteele/comitup #3: .local domain is incomatible with Avahi/ZeroConf on Linux

ringe, 2017-07-27:
My Ubuntu install tells me the Avahi service is disabled as soon as I connect to the comitup hotspot, complaining it is a .local domain.

davesteele, 2017-07-27:
Zeroconf/Avahi/Bonjour client support is not complete across clients (except for Apple, who originated the technology). Google says this about Ubuntu:

https://askubuntu.com/questions/339702/network-service-discovery-disabled-what-does-this-mean-for-me
http://smallbusiness.chron.com/resolving-local-ubuntu-38861.html

Note that the hotspot IP addess is always 10.42.0.1, and that a [Zeroconf Browser](https://apps.ubuntu.com/cat/applications/precise/avahi-discover/) browser can always be used to find the IP address when connected to another access point.

jonassmedegaard, 2018-10-15:
I believe the subject of this issue is not linux being incompatible with Zeroconf, but instead - as is also tracked in [the bugreport](https://bugs.launchpad.net/ubuntu/+source/avahi/+bug/327362) linked in a comment to the askubuntu question linked above - that mDNS is incompatible with DNS-resolvable .local top-level domain.

Zeroconf is multiple things, one of them being multicast DNS resolving, which occupies the .local domain and uses it solely for Link-Local address resolving.
Some internet providers wrongly resolve queries for .local domains.  They shouldn't. That is a bug in the offerings of those ISPs, and a workaround is to use different DNS resolvers - e.g. the famous Google 8.8.8.8 resolver or the newer faster 1.1.1.1 resolver.

Please consider renaming this issue to e.g. ".local mDNS domain is incompatible with .local resolved as conventional non-Zeroconf DNS by some ISPs"

davesteele, 2021-04-10:
The upstream bug has a [mitigation](https://github.com/lathiat/nss-mdns#etcmdnsallow), for the case where your ISP insists on resolving .local.

closed 2021-04-10 by davesteele