davesteele/comitup #6: Pre and post-install questions

arjunmenon, 2017-09-29:
Hey
I got some questions
1. Is it possible to disable the SSH functionality when connection to temporary hotspot, so that only setting up the home access point functionality is available?
2. A way to set the `commitup-<nnnn>` default password so that it isnt connected by someone else accidently.
3. Does it support Enterprise wireless, where generally it requires a username as well with the password? I see that the `confirm.html` only has the field for password

Some feature request
1. Change the `commitup-<nnnn>` to `commitup-<ipaddr>`. This would make it easy for the end user to know the ip addr to navigate straight up witouth a zeroconf browser.

davesteele, 2017-09-30:
1. Internally, there are callbacks upon entering each connection state. I don't intend to support SSH-by-state, but I would consider exposing the callbacks, as a feature request.
2. I suppose the web interface could do that, but it could be a huge security hole. Perhaps if it is available only if the pw is default. This would be the first explicit Pi feature of the software.
3. No. I've not had access to such a device to debug the capability.

Re FR1 - The goal is for the name to be somewhat persistent, and to be common across interfaces.

arjunmenon, 2017-09-30:
Hey
I am facing some installation issues
```bash
pi@raspberrypi:~ $ sudo apt-get update
Get:1 http://davesteele.github.io comitup-jessie InRelease [3,919 B]                                            
Ign http://davesteele.github.io comitup-jessit InRelease                                                        
Hit http://mirrordirector.raspbian.org jessie InRelease                                          
Hit http://archive.raspberrypi.org jessie InRelease                        
Ign http://davesteele.github.io comitup-jessit Release.gpg                       
Get:2 http://davesteele.github.io comitup-jessie/main armhf Packages [553 B]     
Ign http://davesteele.github.io comitup-jessit Release                                                          
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages                
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages                                   
Hit http://archive.raspberrypi.org jessie/main armhf Packages                                          
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages               
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages                                                
Hit http://archive.raspberrypi.org jessie/ui armhf Packages                                                     
Ign http://davesteele.github.io comitup-jessie/main Translation-en_GB                                           
Ign http://davesteele.github.io comitup-jessie/main Translation-en                                              
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB                                         
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en                                            
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB                                 
Err http://davesteele.github.io comitup-jessit/main armhf Packages
  404  Not Found [IP: 151.101.37.147 80]
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en  
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB                         
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB                                   
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en                                           
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB                                             
Ign http://archive.raspberrypi.org jessie/ui Translation-en                                      
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en   
Ign http://davesteele.github.io comitup-jessit/main Translation-en_GB
Ign http://davesteele.github.io comitup-jessit/main Translation-en
Fetched 4,472 B in 23s (188 B/s)
W: Failed to fetch http://davesteele.github.io/comitup/repo/dists/comitup-jessit/main/binary-armhf/Packages  404  Not Found [IP: 151.101.37.147 80]

E: Some index files failed to download. They have been ignored, or old ones used instead.
```
Through PPA installation fails. Had to manually wget the deb package, build the dependencies and the software.

But, however, nothing happens after reboot. I had removed my network settings from `/etc/wpa_supplicant/wpa_supplicant.conf`
I did not see the `commit-up` AP in the wifi list.

davesteele, 2017-09-30:
It looks like you misspelled Jessie in sources.

arjunmenon, 2017-09-30:
Hey
Yeah I had created a separate source list earlier. Commented that out.

But comit-up it is not running. This was in the default interfaces 
```# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

allow-hotplug wlan1
iface wlan1 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
Should it be like this
```
auto lo 
iface lo inet loopback
iface eth0 inet dhcp 

allow-hotplug wlan0

iface wlan0 inet static
  address 192.168.10.1
  netmask 255.255.255.0
```
I have removed the wifi info from `wpa_supplicant`. comit-up does not show up as AP?

davesteele, 2017-09-30:
Comitup pulls in and use network manager. There should be no references to the wlan devices in interfaces.

arjunmenon, 2017-09-30:
Like this
```
auto lo 
iface lo inet loopback
iface eth0 inet dhcp 

allow-hotplug wlan0
```

davesteele, 2017-09-30:
That should work. The hotplug line isn't needed.

arjunmenon, 2017-09-30:
Okay. I would remove the last line as well and hook it back to my TV now. Be right back.

arjunmenon, 2017-09-30:
Alright :grinning: 
This looks like the most robust wifi configuration out here. I have tried dozens before this and most fail for one reason or another. This looks quite robust!

It also auto-detects, quickly, if the network changes connection or params. Brings it up ASAP. Cool!
So thanks.

davesteele, 2017-09-30:
I'm glad it is working for you. Enjoy.

arjunmenon, 2017-09-30:
Brimming with ideas-
1. Captive portal?
2. Force soft/hard reset. Direct command to `comitup-cli > (d)`?
3. Introduce AP passkey - Important.

davesteele, 2017-09-30:
Patches welcome.

davesteele, 2017-10-01:
Concerning comitup-cli ? (d). This is essentially 'nmcli connection delete id <ssid>'. Not tested, but it should work.

arjunmenon, 2017-10-02:
Thanks I would test that out. Concerning patches, sure. I will start looking into the code. I am more interested in the default password setup.

arjunmenon, 2017-11-20:
Hey
If I can ask, can this software have a more liberal license? Or a dual license atleast, based on your terms? 
This is a good software.

davesteele, 2017-11-20:
At first blush, I don't see a need for it. If you use it in your product, the license only covers the daemon.

arjunmenon, 2018-01-23:
Hey
If comitup is triggered and the wifi is also backup, will it make an attempt to reconnect to it?
As of now I have noticed it does not. Is there a behavior which can be set?

davesteele, 2018-01-23:
The periodic timeout in HOTSPOT mode queries NetworkManager for local Access Points, compares that list to configured connections, and attempts to connect to something in the resulting list.

NetworkManager is not always a reliable source for this, particularly in AP mode, so "iw dev scan" is also used. I also temporarily kill the AP hotspot in the timeout (only if no-one is connected to the comitup hotspot), if no APs have been seen.

Are you using the 3/W internal wifi for the hotspot? Do you have two wifi interfaces?

If the problem is no visibility of local APs while in HOTSPOT mode, perhaps Comitup should start in CONNECTING mode with a seeded list.

arjunmenon, 2018-01-24:
> Are you using the 3/W internal wifi for the hotspot? Do you have two wifi interfaces?
I dont know what 3/W is?

I have only one wifi router. So when comitup is up, it does not go back to connecting to my router if it is back up.

Since timeout is set at 5 sec, I have waited longer. It does not reconnect to my router.

davesteele, 2018-01-24:
Raspberry Pi 3/Raspberry Pi Zero W.

By interfaces, I mean wifi interfaces on the Pi.

5 seconds is a one-time-only startup deal. Normally, it is [3 minutes](https://github.com/davesteele/comitup/blob/4d9b1ac07c7c6284d2e74d2cc6d9e83f3e150a47/comitup/states.py#L262).

It is more aggressive about the connection attempt if [no-one has connected](https://github.com/davesteele/comitup/blob/4d9b1ac07c7c6284d2e74d2cc6d9e83f3e150a47/comitup/states.py#L141) to the Comitup hotspot, or if you have two wifi interfaces on the Pi.

arjunmenon, 2018-01-24:
OK.
In my opinion, if it finds existing wifi router again, it should connect immediately.
Only if the comitup is manually triggered, pushing a button or from cmd, it should wait for 180 sec. That should give sufficient time for a user to enter new wifi details, else try it from the conn_list.

davesteele, 2018-01-24:
The timeout is only invoked after it has decided there is nothing to connect to.

I'm willing to consider patches. Make sure you test them well first.

closed 2018-04-09 by davesteele

theronic, 2018-12-02:
Does comitup reconnect to the previously configured Wi-Fi access point when that access point went away and came back? I'm seeing that comitup does *not* reconnect when the surrounding wi-fi went down.

davesteele, 2018-12-02:
Yes, it should.