davesteele/comitup #7: Pi zero W not supported ?

pagaille, 2018-03-10:
Hi,

As far as I can see the Pi zero W is not supported. The AP just don't show up, even one without SSID.

Is there anything we can do ?

Thanks

davesteele, 2018-03-10:
I'm using it on a Zero W right now without issue. What version are you using? Did you use the SD card image, or install the package?

pagaille, 2018-03-10:
Thanks for your answer. I tried both. I checked that /etc/interfaces was empty. How can I help debug the issue ?

davesteele, 2018-03-10:
Try running 'comitup-cli' as root, and see what that tells you ('[man comitup-cli](https://davesteele.github.io/comitup/man/comitup-cli.1.html)'). Also check the comitup log in /var/log.

If I'm doing a lot of debugging, I kill the service ('sudo systemctl stop comitup' and 'comitup-web') and run it from the command line ('sudo /usr/sbin/comitup').

pagaille, 2018-03-10:
/var/log showed` comitup - WARNING - Error connecting hotspot`
I'll investigate further but sadly I only can access it through wifi since I don't have a HDMI monitor right now. That complicates things a bit ;-)

pagaille, 2018-03-10:
Long story short, it works now.

I tried installing from scratch using the latest packages, comitup_1.2.3-1_all.deb and python3-networkmanager_2.0.1-4_all.deb using gdesbi to resolve the dependencies.

I don't know why the image doesn't work. No time to test again right now.

THANKS for that really good and clever software !

davesteele, 2018-03-10:
That warning indicates an error with NetworkManager, which, I believe, is mitigated with python3-networkmanager 2.0.1-4. That package is included on the comitup repo, and is pulled in by the lastest comitup version.

closed 2018-04-09 by davesteele