davesteele/comitup #8: Bug when choosing (i)nfo in comity-cli

pagaille, 2018-03-11:
```
$ comitup-cli

State: CONNECTED
Connection: MattLabs
Available commands:
    (i)nfo
    (r)eload
    (d)elete connection
    (q)uit
command?: **i**
Traceback (most recent call last):
  File "/usr/sbin/comitup-cli", line 11, in <module>
    load_entry_point('comitup==1.2.3', 'console_scripts', 'comitup-cli')()
  File "/usr/share/comitup/cli/comitupcli.py", line 115, in interpreter
    commands[cmd].fn(connection)
  File "/usr/share/comitup/cli/comitupcli.py", line 41, in do_info
    info = ciu.ciu_info(connection)
AttributeError: module 'comitup.client' has no attribute 'ciu_info'
```

pagaille, 2018-03-11:
Scratch that. It looks like comitup doesn't like [adafruit's read-only raspberry](https://learn.adafruit.com/read-only-raspberry-pi/overview) script, even when starting in read-write mode (using a GPIO jumper) : the DHCP server doesn't work anymore in hotspot mode.
I wonder if installing comitup **after** having launched the read-only script (and having boot in RW mode of course).

davesteele, 2018-03-11:
Bug confirmed. unrelated to the read-only script.

pagaille, 2018-03-12:
I guess it has nothing to do with the DHCP server not working correctly ?

davesteele, 2018-03-12:
That particular problem? No.

I wouldn't be surprised about problems related to the read-only script. It depends mostly on how it handles /var/log files. 

I'd run comitup once before making things read-only. If the /var/lib file is unwritable on the first run, it will crash. If it is transient, the name of the hotspot will change on every boot.

pagaille, 2018-03-23:
Txs Dave.

That's what I did. comitup was installed and tested before I ran the script. 

There are some comments here and there regarding the DHCP server not being happy on a read-only system. https://www.raspberrypi.org/forums/viewtopic.php?t=62705

davesteele, 2019-02-25:
See https://github.com/davesteele/comitup/issues/25#issuecomment-467087333.

davesteele, 2019-03-04:
Update, based on the current master.

Comitup and NetworkManager log data. A ram mount of /var/log is recommended.

NetworkManager stores new connection information under /etc, and Comitup creates a /var/lib/comitup file on first run. You would need to establish the desired connections before locking down the file system.

Comitup now manages the dnsmasq-based DNS server. It needs to be able to regularly write in /var/run, and to the /var/lib/comitup directory (DHCP lease information).

There may be more - I haven't actually tried it.

closed 2020-06-17 by davesteele