davesteele/comitup #10: SSH not responding after setup wifi.

Webonaute, 2018-05-29:
when setting up the wifi on http://raspberrypi.local, the ssh stop responding after.

I reconnect to my router wifi, I can see the device connect to 192.168.1.113. I can ping the device but it not answer to ssh. how do I fix that?

davesteele, 2018-05-29:
What is the date in the name of the image file?

Did you verify that it is the Pi that you are talking to?

Could there be a firewall issue between you and the Pi?

Webonaute, 2018-05-29:
2018_05_24 is the date,

I have no firewall. I can connect to any other device I have over ssh.
Yes. if I connect the keyboard/monitor to the Pi, I see the right IP.

I can see the wifi is configured on wlan0, I can also run ssh localhost on the PI and sshd is responding.
From the Pi, I can ping my mac or google.com.

Webonaute, 2018-05-29:
netstats -na |grep :22 show me that the port is open on any ip on port 22.

Webonaute, 2018-05-29:
seem that ssh respond only to pi@raspberrypi dns

davesteele, 2018-05-29:
I probably won't have a chance to look at this before next week. 

I'll say this - that version was very lightly tested. If you want to address this before then, you may want to try the [previous release](http://steele.debian.net/comitup/).

Webonaute, 2018-05-29:
np. I get around by connecting directly with monitor and keyboard and now that I know it work with dns, its not a big problem. it's just annoying when its first time you setup a PI. it took me lot of time to figure it out. I tried your image because I could not make any other raspbian image working on my sd card. (probably my SD card reader who is really old, IDK.) there where not booting. than I try your image and it boot. but had network issue. lol. anyway. now I can start testing all those cool sensor from the kit I buy! :)

davesteele, 2018-06-16:
Just tried it out. Could not replicate the problem.

closed 2018-06-16 by davesteele