davesteele/comitup #11: Events

arjunmenon, 2018-06-04:
Hey
I have certain scripts that I would like to run when comitup successfully establishes wifi connection. Is it possible to have comitup run them? I dont want to use rc.local or crontab or other ways. I was looking to make it depended on comitup.

In addition to comitup starting scripts on boot, so if wifi temporarily goes out of range, comitup stops those scripts and if the wifi comes back up, comitup re-runs those scripts.

Is it possible?

_PS - comitup running in Raspberry Pi3, with one Ruby script and other a python script_

davesteele, 2018-06-04:
Look at the external_callback parameter in the comitup-conf man page.

cyberchucktx, 2018-07-10:
Is there a more comprehensive FAQ for ComitUp? And is there a Slack or forum for this system/tool?

It is pretty fantastic, I've been hacking the pieces of this kind of system together for awhile; nice work on the part of the author. 

If there is an FAQ I could contribute, and would definitely like to discuss tips & tricks. 

Some FAQs: 
1) What happens if I enter the WiFi password wrong? 
--- Does it show invalid connection or just hang? 
--- In that case, how do I get back to the point where I can put in the correct password? 
2) Can I clear out the known access points in a straightforward manner? If so, how? 
3) If I use NGINX as the "back end" can I easily revert back to another access point assuming I'm in an environment where I have multiple WiFI networks? 

I can dig through the source code to find this out, but if this info is already somewhere I'd be glad to contribute & help with editing it. 

BTW, I'm using this (ComitUp) in my  [Eink Badge Project](https://hackaday.io/project/159554-inkyphat-epapereink-display-for-badges) at Hackaday.io @

davesteele, 2018-07-10:
The [FAQ](https://github.com/davesteele/comitup/wiki/FAQ) is on the wiki (I've added a link to the home page). According to GitHub, any logged in user can modify.

There hasn't been a clamoring for a support forum. 

To answer your questions:
1. If you type the password wrong, it will eventually fail and revert to the HOTSPOT mode. It will retry the connection every couple of minutes, unless someone is connected to the hotspot, in which case it will wait to be commanded. The failure is noted in the log.

2. Known connections are managed by NetworkManager, and can be deleted using features of that package. For instance, you can just delete the appropriate files in /etc/NetworkManager/system, or use nmcli to do it for you more cleanly. I usually use the 'delete' command in comitup-cli. You'll note in the comitup man page there is also a D-Bus call for this.

3. If you have multiple access points that you have configured that are visible at the same time, the AP selection process is undefined. Again, it is just NetworkManager underneath, so you could provide a way to query it and to connect to the desired AP. Comitup **should** tolerate that action, but I haven't tested it.

Looks like an interesting project - glad I could help. You have linked a particular image - note that they tend to disappear in 6 months or so.

cyberchucktx, 2018-07-10:
David: 

I just attempted to update the FAQ on GITHUB; I cloned locally and then pushed back up to the master. 

Sorry, I'm used to using GITHUB locally and frankly haven't used it in  awhile. Hope I didn't mess anything up. 

Charlie

cyberchucktx, 2018-07-10:
David: 

> Looks like an interesting project - glad I could help. You have linked a particular image - note that they tend to disappear in 6 months or so.
I don't understand your point. I downloaded this image to my machine and then pushed back up to the hackaday.io site (it is nota URL but a "local" image to the site).
Or am I misunderstanding you? 

Thanks,Charlie

davesteele, 2018-07-10:
Images are updated about every 3 months (I track upstream), and I keep the last two. So in 6 months, don't be surprised if the "2018-06-29" image disappears.

Try http://steele.debian.net/comitup/. That won't change.

Also, do you want to make Comitup step 1? If you user goes with the img option, he's going to undo the InkyPHAT install.

There's no damage to the Wiki, but I don't see any changes either.

cyberchucktx, 2018-07-10:
David: 

Weird. I don't know why the changes didn't get propagated. 

As I said, used the GitHub Desktop on my Mac, and sometimes that 
system does some weird stuff under the hood. I"ll go back & do it CLI-based so I cansee what's going on.
Here are the additions (questions by me, answers by you): 

--------------------------------
**What happens if I enter the WiFi password wrong?**

If you type the password wrong, it will eventually fail and revert to the HOTSPOT mode. It will retry the connection every couple of minutes, unless someone is connected to the hotspot, in which case it will wait to be commanded. The failure is noted in the log.

**Can I clear out the known access points in a straightforward manner? If so, how?**

Known connections are managed by NetworkManager, and can be deleted using features of that package. For instance, you can just delete the appropriate files in /etc/NetworkManager/system, or use nmcli to do it for you more cleanly. I usually use the 'delete' command in comitup-cli. You'll note in the comitup man page there is also a D-Bus call for this.

**If I use NGINX as the "back end" can I easily revert back to another access point assuming I'm in an environment where I have multiple WiFI networks?**

If you have multiple access points that you have configured that are visible at the same time, the AP selection process is undefined. Again, it is just NetworkManager underneath, so you could provide a way to query it and to connect to the desired AP. Comitup should tolerate that action, but I haven't tested it.

--------------------------------

cyberchucktx, 2018-07-10:
David: 

Ah. When you say "image" you mean "disk image". For some odd reason I was thinking aboutthe "image" on my Hackaday.io page. 

> Also, do you want to make Comitup step 1? If you user goes with the img option, he's going to undo the InkyPHAT install.Whups.  Great feedback, thanks.
Charlie

arjunmenon, 2018-07-19:
Hey
for external_callback
when you say
> It will include a single argument, either 'HOTSPOT', 'CONNECTING', or 'CONNECTED'. 

Where exactly should it be included in the eg given, `external_callback: /usr/local/bin/comitup-callback`

davesteele, 2018-07-19:
The script will be called with that value as the first argument, e.g. "/usr/local/bin/comitup-callback CONNECTING". If that script is a shell script, it is available as the variable "$1".

arjunmenon, 2018-07-19:
Need more clarification.
I have a ruby server which I want to run on CONNECTED

`/home/pi/server/run.rb CONNECTED`

Now it is clear that my script run.rb should account for the args, question is twofold.
1. Why? What is the `run.rb` suppose to do with that args?
2. How will the comitup service suppose to interact with `run.rb`? Is `run.rb` suppose to call any comitup commands? Or comitup is only looking for those key constants and simply figure out which scripts to run based off that? I dont have to do anythin else?

davesteele, 2018-07-19:
1. Decide if it wishes to act based on that type of event, and respond appropriately. Callbacks are not required.
2. Comitup doesn't care about the actions of the callback script. It is for your use. Note that the script should return promptly - comitup processing is suspended while it runs.

arjunmenon, 2018-07-19:
Hey
About note 2
Basically to be safe I should have a script which would trigger other scripts,.? I can have a bash file which can start stop the ruby server, right.

davesteele, 2018-07-19:
Ues

arjunmenon, 2018-07-27:
> In the conf file, the "$1" is not necessary, and likely why your aren't seeing logs (it doesn't log in the case where the file doesn't exist).

"Not necessary" does it also mean that it wouldn't matter if it is present?
It would also be helpful to see atleast some information in comitup logs that the user made an attempt to do something with the callback?

> Regarding the "touch", don't assume that you know what the current working directory is.

Well I did try to do `locate logfile.txt`. No result. Perhaps I can pipe it to system logs then.

> And, oops, there is a bug in comitup that causes a crash if your script is not owned by the root owner and group. Added as #14.

It is owned by root. I even changed the permissions to 777 to be absolutely clear. Still nothing happened.

I think it would be fair to trouble you for a small example ;)

davesteele, 2018-07-27:
Your example is fine - just add a full path for the logfile, and remove " $1" from the callback (comitup is checking for the existence of the file "/home/pi/comit.sh $1", which is not present). The callback code is [here](https://github.com/davesteele/comitup/blob/bab9702029bd4f0c64a04847ecf8a9c1e4a348b0/comitup/statemgr.py#L112). The isfile() call is currently returning false. Notice in subprocess.call() that the command to run is separated into individual elements. The "$1" is not split off of the executable path here, and so is considered part of the script name.

closed 2020-06-17 by davesteele