davesteele/comitup #12: WARNING - Error connecting hotspot

4fd81048a9cd8a3fd43f1f4e52f84181, 2018-06-24:
Hello.

I've done a NOOBS V2.8.1 to install Raspbian Stretch on my Raspberry Pi 3 Model B.

I used the Graphical Preferences Config to change the password, set to boot to gui, enable ssh, set Localization options (Canada English UTF-8), Timezone, Keyboard Layout and WiFi Country (Canada) and rebooted.

Then added `deb http://davesteele.github.io/comitup/repo comitup main` to /etc/apt/sources.list and did apt-get update, apt-get upgrade, apt-get dist-upgrade and rebooted again.

Finally apt-get install comitup and I'm getting this in /var/log/comitup.log

```
2018-06-23 23:02:23,771 - comitup - INFO - Setting state to CONNECTING
2018-06-23 23:02:29,000 - comitup - INFO - Setting state to HOTSPOT
2018-06-23 23:02:29,169 - comitup - INFO - Activating hotspot
2018-06-23 23:02:29,304 - comitup - WARNING - Error connecting hotspot
2018-06-23 23:05:30,421 - comitup - INFO - Setting state to CONNECTING
2018-06-23 23:05:35,648 - comitup - INFO - Setting state to HOTSPOT
2018-06-23 23:05:35,827 - comitup - INFO - Activating hotspot
2018-06-23 23:05:35,962 - comitup - WARNING - Error connecting hotspot
2018-06-23 23:08:37,117 - comitup - INFO - Setting state to CONNECTING
2018-06-23 23:08:42,346 - comitup - INFO - Setting state to HOTSPOT
2018-06-23 23:08:42,551 - comitup - INFO - Activating hotspot
2018-06-23 23:08:42,683 - comitup - WARNING - Error connecting hotspot

```
and no comitup-xxxx wifi is being found.

I've been connecting over the wired interface and eth0 is getting an address from my router. (I just rebooted and disconnected the wired network and getting the same results by checking from the console)

I've also noticed that my interfaces are the old eth0 and wlan0 not the unique ids that some people mention they get with Stretch.

```
pi@comitup:~ $ nmcli
eth0: connected to eth0
        "0424 ec00"
        ethernet (smsc95xx), B8:27:EB:9F:2C:10, hw, mtu 1500
        ip4 default
        inet4 192.168.137.15/24
        inet4 192.168.137.49/24
        inet6 fe80::30b1:4ac5:d564:b93c/64

wlan0: unavailable
        "Broadcom "
        wifi (brcmfmac), 1A:D1:E1:0D:ED:AC, hw

lo: unmanaged
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

DNS configuration:
        servers: 192.168.137.1
        domains: mshome.net
        interface: eth0

Use "nmcli device show" to get complete information about known devices and
"nmcli connection show" to get an overview on active connection profiles.

Consult nmcli(1) and nmcli-examples(5) manual pages for complete usage details.
pi@comitup:~ $ nmcli device
DEVICE  TYPE      STATE        CONNECTION
eth0    ethernet  connected    eth0
wlan0   wifi      unavailable  --
lo      loopback  unmanaged    --
pi@comitup:~ $ sudo nmcli device show wlan0
GENERAL.DEVICE:                         wlan0
GENERAL.TYPE:                           wifi
GENERAL.HWADDR:                         1A:D1:E1:0D:ED:AC
GENERAL.MTU:                            0
GENERAL.STATE:                          20 (unavailable)
GENERAL.CONNECTION:                     --
GENERAL.CON-PATH:                       --
pi@comitup:~ $ sudo nmcli device connect wlan0
Error: Failed to add/activate new connection: A 'wireless' setting is required if no AP path was given.
pi@comitup:~ $ nmcli connection
NAME                UUID                                  TYPE             DEVICE
eth0                610a5933-75ac-4b02-b542-18794fd6bbf9  802-3-ethernet   eth0
Wired connection 1  d943dc02-d073-32c3-a0c8-f61c8e4b894e  802-3-ethernet   --
comitup-3129        c9745511-ee61-4ba8-a4f3-c4437b6eee24  802-11-wireless  --
pi@comitup:~ $ sudo nmcli connection up id comitup-3129
Error: Connection activation failed: No suitable device found for this connection.
pi@comitup:~ $ nmcli connection show comitup-3129
connection.id:                          comitup-3129
connection.uuid:                        c9745511-ee61-4ba8-a4f3-c4437b6eee24
connection.stable-id:                   --
connection.interface-name:              wlan0
connection.type:                        802-11-wireless
connection.autoconnect:                 no
connection.autoconnect-priority:        0
connection.autoconnect-retries:         -1 (default)
connection.timestamp:                   0
connection.read-only:                   no
connection.permissions:
connection.zone:                        --
connection.master:                      --
connection.slave-type:                  --
connection.autoconnect-slaves:          -1 (default)
connection.secondaries:
connection.gateway-ping-timeout:        0
connection.metered:                     unknown
connection.lldp:                        -1 (default)
802-11-wireless.ssid:                   comitup-3129
802-11-wireless.mode:                   ap
802-11-wireless.band:                   --
802-11-wireless.channel:                0
802-11-wireless.bssid:                  --
802-11-wireless.rate:                   0
802-11-wireless.tx-power:               0
802-11-wireless.mac-address:            --
802-11-wireless.cloned-mac-address:     --
802-11-wireless.generate-mac-address-mask:--
802-11-wireless.mac-address-blacklist:
802-11-wireless.mac-address-randomization:default
802-11-wireless.mtu:                    auto
802-11-wireless.seen-bssids:
802-11-wireless.hidden:                 no
802-11-wireless.powersave:              default (0)
ipv4.method:                            shared
ipv4.dns:
ipv4.dns-search:
ipv4.dns-options:                       (default)
ipv4.dns-priority:                      0
ipv4.addresses:
ipv4.gateway:                           --
ipv4.routes:
ipv4.route-metric:                      -1
ipv4.ignore-auto-routes:                no
ipv4.ignore-auto-dns:                   no
ipv4.dhcp-client-id:                    --
ipv4.dhcp-timeout:                      0
ipv4.dhcp-send-hostname:                yes
ipv4.dhcp-hostname:                     --
ipv4.dhcp-fqdn:                         --
ipv4.never-default:                     no
ipv4.may-fail:                          yes
ipv4.dad-timeout:                       -1 (default)
ipv6.method:                            ignore
ipv6.dns:
ipv6.dns-search:
ipv6.dns-options:                       (default)
ipv6.dns-priority:                      0
ipv6.addresses:
ipv6.gateway:                           --
ipv6.routes:
ipv6.route-metric:                      -1
ipv6.ignore-auto-routes:                no
ipv6.ignore-auto-dns:                   no
ipv6.never-default:                     no
ipv6.may-fail:                          yes
ipv6.ip6-privacy:                       -1 (unknown)
ipv6.addr-gen-mode:                     stable-privacy
ipv6.dhcp-send-hostname:                yes
ipv6.dhcp-hostname:                     --
ipv6.token:                             --
proxy.method:                           none
proxy.browser-only:                     no
proxy.pac-url:                          --
proxy.pac-script:                       --
```

Any ideas?  It looks like wlan0 is not available to NetworkManager?

4fd81048a9cd8a3fd43f1f4e52f84181, 2018-06-24:
I believe the problem is wpa_supplicant.conf.  I renamed it out, rebooted and it's working.  Probably the WiFi country setting I did at the beginning.

When I boot with the .conf
```
pi@comitup:/etc/wpa_supplicant $ ps -ef | grep wpa
root       406     1  0 00:01 ?        00:00:00 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0 -Dnl80211,wext

```
and when I reboot after renaming it
```
pi@comitup:~ $ ps -ef | grep wpa
root       469     1  0 00:05 ?        00:00:00 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
```

I'm not able to find the correct service(s) to restart or kill to get NetworkManager to take it, a reboot gets it though.

closed 2018-06-24 by 4fd81048a9cd8a3fd43f1f4e52f84181

davesteele, 2018-06-24:
Thanks for adding the resolution. I've added it to https://github.com/davesteele/comitup/wiki/Installing-Comitup.