davesteele/comitup #13: New version issue

arjunmenon, 2018-07-19:
Hey
The new version is introducing issues in installation in Rpi Jessie
Jessie doesn't have python 3.5 yet. Only python 3.4
Manually installing Python 3.5 is not recognized by comitup.

davesteele, 2018-07-19:
It should run with 3.4 - I just haven't thought about Jessie for a while. You may be able to force the install, but I'd really recommend updating to stretch (or stick to an [older version](https://davesteele.github.io/comitup/archive.html) of comitup).

Your manual install of Python 3.5 didn't affect the Debian package status for Python, so the package installer ignored it.

arjunmenon, 2018-07-19:
Hey
I wont prefer an dist upgrade at the moment. I have lot  of packages installed and working and in sync with each other. It is stable at present. Dont want to break that.

I have installed comitup 1.1.2 to get that external_callback feature. Earlier I had 0.9

closed 2020-06-09 by davesteele