davesteele/comitup #16: Using Assets on html files

krebbi, 2018-08-14:
Hi,

great package! 

I'm just wondering how to integrate some css and other assets to the template to pretty up the pages?

davesteele, 2018-08-14:
The web interface is a flask app, installed to /usr/share/comitup/web. Links to assets can be added to the html templates, and provisions made to flask to serve the new content.

I'd welcome patches to pretty up the web pages.

krebbi, 2018-08-14:
so could I just make some more nice pages and leave the flask-stuff at your side?

davesteele, 2018-08-14:
I don't understand the question.

krebbi, 2018-08-19:
just created a PR #18 with prettified Webpages.

While testing I figured out that on the package installed version you write into the /web/comutupweb.py an "HelloWorld" Function which I cannot find in this repo. 
So hopefully my change in the repos comitupweb.py will not break your package. 

howevery, just let me know if you like the PR.

closed 2019-01-18 by krebbi