davesteele/comitup #17: Captive Portal functionality

krebbi, 2018-08-14:
How difficult would it be to integrate a real captive portal functionality to this package like descriped here?

https://en.wikipedia.org/wiki/Captive_portal#Implementation

davesteele, 2018-08-14:
I'm not sure what you are asking. If you want to implement a captive portal on the Pi, that's not a Comitup question. Once the outbound connection is made, the comitup web interface is taken down, and replaced with [web_service](https://davesteele.github.io/comitup/man/comitup-conf.5.html), if defined. Just point that to a captive portal implementation (which I imaging exists somewhere).

If you want to automatically bypass a captive portal, it would be hard to make a general solution for that.

krebbi, 2018-08-14:
When connecting with a device to the running comitup, it would be nice to get the webpage from comitup without the need to open the browser and go to 10.41.0.1
to have it a bit more userfriendly. ;-)

davesteele, 2018-08-14:
Ah. It would be fairly easy to do a redirect for http traffic. For https, it's not really practical, unless your device (e.g. cell phone) has some special logic for it already. It would be interesting to try.

krebbi, 2018-08-15:
Maybe this could help you?

https://blog.trifork.com/2013/01/15/building-a-captive-portal-controlling-access-to-the-internet-from-your-network/

cyberchucktx, 2018-08-15:
Well, not really. adding all this to the comitup system would be (IMHO) unecessarily complex. I do agree that a reformatting of the basic WiFi page can be done rather easily (first).

krebbi, 2018-08-25:
Haven't had the chance to test this as I'm on vacation this and the next week. But I'm looking forward giving it a try a soon as I'm back at my Pi.

davesteele, 2018-08-26:
The captive-portal branch has a web page that comes up correctly, but doesn't yet totally work with Android or IOS. I'm not getting the config parameters from Android for some reason, and IOS needs the ICMP *-unreachable ICMP filters added back into iptmgr. Let me know if you have any luck with that.

krebbi, 2018-08-28:
I'm still 6 days away from being back at my desk, but I just found this while researching the topic:

https://tools.ietf.org/html/rfc7710

Hopefully it will assist you.

davesteele, 2018-08-28:
Interesting. I'm just hijacking external web requests. That part is
working- phones detect the login requirement. It's the login in part that
is the problem.

krebbi, 2018-09-04:
Okay, I figured out that there are several issues. Primary at my iOS Device the Captive Portal Page doesn't show up automatically after connecting to the comitup-XXXX Hotspot. 
But it show when trying to reach any website. Trying to visit google.com brings up the Comitpu-Website. But the URL in the addressbar is still "google.com". 
Then choosing the WiFi and entering the key works like expected.

But there is a little mess in all of the python files regarding the version. Some files need 2.7 and others 3. 
When having the wrong default version set up, it doesn't work. the networkmanager doesn't work properly. and setting things up via comitup-cli doesn't work either.

So I would suggest you'll try to get all files and dependencies to one version (best 3.5). 

I'll go thtough the ieft-paper and check if I can adapt that into comitup.

davesteele, 2018-09-05:
On IOS, I'm seeing the behavior I saw before blocking outbound ICMP unreachables - the AP connection is dropped after a couple of seconds (6ce7c1c94).

Where are you seeing a python 2.7 requirement? I had an obsolete dependency on the configparser deb, but I don't see anywhere 2.7 being called out.

krebbi, 2018-09-05:
Hmm.. strange.. 

Just wrote a new B+ SD-Card and started over.
After cloning the repo and runninig python3 setup.py install I get:

dnslib is missing (solved with sudo pip3 install dnslib)
networkmanager is missing (sudo pip3 install python-networkmanager)
the name org.freedesktop.networkmanager was not provided by any .service files (sudo apt-get install network-manager)

And now I'm ending up with 
"dbus.exceptions.DBus.Exception: org.freedesktop.systemd1.NoSuchUnit: Unit comitup-web.service not found."

Any idea what went wrong?

davesteele, 2018-09-05:
The dnslib requirement is new with the captive portal. I think I made a local copy to deal with a missing deb. I'm not near the code right now.

For the rest, I use deb installs. Look at the dependencies in the debian/control file in the debian branch. That is also where systemd service files are set up. You can work around this by installing an old comitup deb, then deploying new code by copying to /usr/share/comitup. Or shortcut by starting with the Comitup image.

davesteele, 2018-09-05:
dnslib should be [this one](https://bitbucket.org/paulc/dnslib/).

krebbi, 2018-09-05:
okay, its getting worse and worse.. nothing works anymore. exceptions over exceptions. 

next SD card....

davesteele, 2018-09-05:
The more the merrier. I'll see about ensuring the repo is matched to my latest by Friday EOB a the latest.

davesteele, 2018-09-07:
Trivial change pushed. I'll be packaging dnslib separately for stretch installs. Pip on the Pi for now.

davesteele, 2019-03-04:
Captive portal implementation is in master.

closed 2019-03-04 by davesteele

krebbi, 2019-03-21:
Is it available over your PPA repository already?

davesteele, 2019-03-22:
No, it is not ready for prime time yet.

commit fff0a7d4 (2019-03-04): Add captive portal to changelog

Closes #17.