davesteele/comitup #20: Retain Access Point and Zeroconf During & After Connection SSID Connection

theronic, 2018-11-06:
Hi Dave, thanks for comitup. A painpoint I'm experiencing with comitup is a Raspberry Pi connecting to a Wi-Fi hotspot where the Internet is broken and becoming unreachable with no recourse to move off that SSID. It would be preferable to retain.

 1. Can you retain the hotspot while connecting to an access point? (I know it's possible if you use the same Wi-Fi channel as target network)
 1. Can you retain the Zeroconf name after connecting?
 1. Can I tell comitup to revert to hotspot mode (or just keep it around), when some health check fails?

Thanks!

(Sidenote: the Tutorial/FAQ do not describe how to start comitup after installing via apt-get)

davesteele, 2018-11-06:
If the upstream AP is unavailable, Comitup reverts to hotspot mode.

1. Yes. This requires a second WiFi adapter.
2. The Zeroconf name is retained on all interfaces across modes.
3. This is the default behavior. You can delete the AP configuration using comitup-cli, or by just removing the config from NetworkManager.

There should be no special startup required after apt-get install (except to insure that the wifi interface(s) are not claimed in /etc/network/interfaces).

davesteele, 2018-11-06:
Package install is covered on the [home page](https://davesteele.github.io/comitup/), and in the [wiki](https://github.com/davesteele/comitup/wiki/Installing-Comitup).

krebbi, 2018-12-07:
@theronic do you mean that for example if the google DNS at 8.8.8.8 is not pingable but a connections is established comitup should switch back to hotspot mode?

closed 2020-06-17 by davesteele