davesteele/comitup #22: Turn on wifi connectivity only when specify pin state is high

venkatyaragalla, 2018-12-17:
I would like to turn on **comitup WI-FI connectivity** only when **raspberry pi pin 18 is HIGH** other wise **COMITUP WI-FI connectivity turn off**. Is it any possible?

davesteele, 2018-12-17:
Comitup is a systemd service, which also uses the NetworkManager service. You could write another that turns them on and off based on the GPIO pin.

venkatyaragalla, 2018-12-17:
Thank you ,Could you please help us where to implement in which file.

davesteele, 2018-12-17:
Well, the systemd configuration files that define the services are in /lib/systemd/system/ (try 'dpkg --listfiles comitup | grep system').

Any new service would need to start, probably before these two, monitor the GPIO, and enable/disable and start/stop those services. You'll need to read up on systemd and GPIO interfaces. It will be a bit of work first time through.

closed 2019-01-26 by venkatyaragalla