davesteele/comitup #23: Order of multiple WLAN devices

krebbi, 2019-01-18:
In the readme is stated

-----
If two wifi interfaces are available, the first will persistently remain the hotspot, and the second will get the external connection. When both are connected, forwarding and masquerading are enabled so that hotspot-connected devices can access external networks.
-----

but here it's exactly the other way around:

wlan0 (build in raspi wifi) connects to choosen network
wlanwlx74da... (USB wifi dongle) seems to be the hotspot

but i cannot see an open hotspot. something is strange here.... 

is there a way to choose the "primary wlan device" in comitup/nm somehow?

davesteele, 2019-01-18:
Not currently.

The assignment of  "ap_device" and "link_device" is done in [modemgr.py](https://github.com/davesteele/comitup/blob/master/comitup/modemgr.py).

What is the make and model of your usb wifi adapter?

closed 2019-01-19 by davesteele

krebbi, 2019-01-19:
thanks a lot!

commit 1e16fd06 (2019-01-19): Allow the primary WiFi interface to be specified

Fixes #23