davesteele/comitup #24: Comitup reverts to hotspot mode if hidden SSID goes away and does not reconnect

theronic, 2019-02-18:
After configuring Comitup to successfully connect to a hidden SSID, Comitup seems to discard its configuration if the hotspot goes away for a while, and does not reconnect.

Is there a way to pre-empt comitup, to always connect to a hotspot if it's available?

Alternatively, can I force nmcli to connect with a cronjob every N minutes?

theronic, 2019-02-18:
Apparently this is a [known bug in NetworkManager](https://bugs.launchpad.net/ubuntu/+source/network-manager-applet/+bug/1542733), which unfortunately affects comitup in a terrible way for remote deployments. We've had to do call-outs as a result of this.

Seems reasonable for Comitup to address this internally with a manual reconnect attempt to hidden SSIDs?

davesteele, 2019-02-18:
An interesting idea. Currently, it goes through all of the available APs reported by NetworkManager, and attempts to connect. This has proved to be problematic - getting NM to see and report the visible APs is a chore. Perhaps comitup should go through the configured AP connections instead.

Your cron idea may work as a stopgap. It depends on how comitup manages/detects the unexpected transition.

commit c4365da5 (2019-03-02): Return all cons as candidates, not just available

NM is unreliable for enumerating available connections, and the old
strategy fails for hidden SSIDs.

Fixes #24.
closed 2019-03-02 by davesteele