davesteele/comitup #27: Changing the SSID

augisbud, 2019-02-25:
Hello, I have been wondering how do I change the SSID for eg. `comitup-<nn>` -> `Hello`.

cyberchucktx, 2019-02-25:
Um, this is clearly in the configuration file (see below). 

#
# Comitup configuration
#

# ap_name
#
# This defines the name used for the AP hotspot, and for the ZeroConf
# host name. The "<nnn>" string, if present, will be replaced with an
# instance-unique, persistent number of the same length. There may be
# up to four "n's" in the string.
#
# ap_name: comitup-<nn>

augisbud, 2019-02-25:
Where can I find the config if I do a package install?

cyberchucktx, 2019-02-25:
You can find the configuration file in the CONF folder on GitHub archive site ( https://github.com/davesteele/comitup ), conf folder at https://github.com/davesteele/comitup/tree/master/conf

augisbud, 2019-02-25:
I meant in linux, raspbian, when I install it using sudo apt-get install comitup

cyberchucktx, 2019-02-25:
To list installed packages on any Debian system (including Raspian), use

apt list --installed

To see where disk paths are related to a given package: 

dpkg -L <packagename-returned-by-apt-list>

davesteele, 2019-02-25:
... which will tell you that the file is /etc/comitup.conf.

"man comitup" would have gotten you there as well (it's actually documented in the comitup.conf man page).

closed 2019-02-26 by augisbud