davesteele/comitup #29: apt-get gives an old version.

augisbud, 2019-03-04:
Hello, when I use the image provided I get totally different design, than when I use apt-get install comitup. Any possibility to change that?

davesteele, 2019-03-04:
Design?

If you mean web design, you can do an apt-get upgrade on the booted image to get the most recent version.

augisbud, 2019-03-04:
I did that, the design hasn't changed, I will try again

augisbud, 2019-03-04:
UPDATE: did sudo apt-get update && sudo apt-get upgrade, the design hasn't changed.

davesteele, 2019-03-04:
What comitup versions are we talking about?

augisbud, 2019-03-04:
the latest i'm getting from sudo apt-get install comitup on raspbian stretch

augisbud, 2019-03-04:
How can I install directly by using git clone, and the setup.py file?

davesteele, 2019-03-04:
Ah. You need to add a line to /etc/apt/sources.list referencing the comitup repository to get the latest:

    deb http://davesteele.github.io/comitup/repo comitup main

Or you can get the latest deb from the [archive](https://davesteele.github.io/comitup/archive.html).

augisbud, 2019-03-04:
still not getting it :(

davesteele, 2019-03-04:
The command to install via setup.py is "python3 setup.py install". 

N.B. I've never done that.

The git archive currently builds the latest deb, using the "debuild" command.

Re - still not getting it. Make sure you do the "apt-get update" after updating sources.list.

augisbud, 2019-03-04:
```
pi@raspberry:~ $ sudo apt install ./comitup_1.3.1-1_all.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'comitup' instead of './comitup_1.3.1-1_all.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 comitup : Depends: python3-networkmanager (>= 2.0) but it is not installable
E: Unable to correct problems, you have held broken packages.

```
output from the deb installation.

YUP I'm dumb, forgot to sudo apt-get update

closed 2019-03-04 by augisbud

davesteele, 2019-03-04:
Re installing from git. For development, what I do is install the deb, then use a small deploy script that copies py files from the cli, comitup, and web directories to the corresponding directories in /usr/share/comitup/, then restart the comitup service.