davesteele/comitup #31: Ethernet connectivity

augisbud, 2019-03-07:
Hello, is it possible to enable ethernet connectivity and if yes, how do I do that?
Because right now ethernet doesn't get an ip from dhcp

davesteele, 2019-03-07:
If you downloaded the Comitup disk image, Ethernet is configured as 10.0.0.2/24. I tried adding a DHCP alias, but it didn't work.

augisbud, 2019-03-12:
If i get it right, the interfaces that are not defined in /etc/network/interfaces are managed by python-networkmanager?

davesteele, 2019-03-12:
Yes, which is in turn being called by Comitup.

augisbud, 2019-03-12:
How do you tell network manager to connect to wifi, create a hotspot etc

augisbud, 2019-03-12:
`nmcli device status
DEVICE  TYPE      STATE         CONNECTION
wlan0   wifi      connected     11111
eth0    ethernet  disconnected  --
lo      loopback  unmanaged     --
`
ok, so this is what I get when I plug in an ethernet cable

augisbud, 2019-03-12:
changing /etc/network/interfaces to 
```
auto lo
iface lo inet loopback
```
enables ethernet, however the wifi ap doesn't start or it doesn't connect to the last known network when ethernet was disconnected

davesteele, 2019-03-12:
The change you made to interfaces should have no ill effect on Comitup (unless you have a wlan* entry), though I don't see input there to have it handle Ethernet.

There is a comitup.log file in /var/log. Check that. Also, you can shut down the service ("systemctl stop comitup") and then run "comitup" from the command line to see stdout output.

augisbud, 2019-03-18:
everything seems to be fine, I'm wondering, how can we disable the hotspot if there is internet from eth0

davesteele, 2019-03-18:
That's not something that would be on my short list.

You might be able to address the underlying issue by adding a password to the hotspot (supported in the next release). Otherwise, you would need a process that detected the Ethernet connection, terminated Comitup, and terminated the hotspot NetworkManager connection.

Kegmaster, 2019-03-19:
I second this.  It would be nice to have the option to disable the hotspot if Internet/Ethernet connectivity was established.

davesteele, 2019-03-19:
I'll consider patches.

davesteele, 2020-06-17:
Ethernet configuration is updated. See the home page.

closed 2020-06-17 by davesteele