davesteele/comitup #35: Not connect to WPA2 networks

iv-med, 2019-04-28:
Not connect to WPA2 networks - not see the lock icon and it connects to it like these networks are open.

davesteele, 2019-04-29:
Could you run "sudo iw dev wlan0 scan" (or whatever the appropriate wifi device is) on the device, and send the portion related to the AP in question?

Comitup looks for a "WPA:" or "RSN" section in that output.

As a workaround, note that the web interface accepts e.g. "https://connect.html?ssid=<ssid>&password=<pw>". Could you try that, to verify that authentication is possible?

davesteele, 2019-04-29:
Note that I have updated the previous comment.

iv-med, 2019-04-29:
> Could you run "sudo iw dev wlan0 scan" (or whatever the appropriate wifi device is) on the device, and send the portion related to the AP in question?
pi@raspberrypi:~ $ sudo iw dev wlan0 scan
BSS 14:4d:67:03:50:c4(on wlan0) -- associated
	TSF: 0 usec (0d, 00:00:00)
	freq: 2412
	beacon interval: 100 TUs
	capability: ESS Privacy ShortSlotTime (0x0411)
	signal: -22.00 dBm
	last seen: 0 ms ago
	SSID: 12343
	Supported rates: 1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 
	DS Parameter set: channel 1
	ERP: <no flags>
	Extended supported rates: 24.0 36.0 48.0 54.0 
	HT capabilities:
		Capabilities: 0x186e
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT RX MCS rate indexes supported: 0-15
		HT TX MCS rate indexes are undefined
	HT operation:
		 * primary channel: 1
		 * secondary channel offset: above
		 * STA channel width: any
		 * RIFS: 0
		 * HT protection: no
		 * non-GF present: 0
		 * OBSS non-GF present: 0
		 * dual beacon: 0
		 * dual CTS protection: 0
		 * STBC beacon: 0
		 * L-SIG TXOP Prot: 0
		 * PCO active: 0
		 * PCO phase: 0
	RSN:	 * Version: 1
		 * Group cipher: CCMP
		 * Pairwise ciphers: CCMP
		 * Authentication suites: PSK
		 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
	WMM:	 * Parameter version 1
		 * BE: CW 15-1023, AIFSN 3
		 * BK: CW 15-1023, AIFSN 7
		 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
		 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec
	WPS:	 * Version: 1.0
		 * Wi-Fi Protected Setup State: 2 (Configured)
		 * Response Type: 3 (AP)
		 * UUID: 11223344-5566-7788-99aa-144d670350c0
		 * Manufacturer: Realtek Semiconductor Corp.
		 * Model: RTL8xxx
		 * Model Number: EV-2010-09-20
		 * Serial Number: 123456789012347
		 * Primary Device Type: 6-0050f204-1
		 * Device name: RTL8196d
		 * Config methods: Ethernet, Label, PBC
		 * RF Bands: 0x3
BSS c0:25:e9:ce:d2:82(on wlan0)
	TSF: 0 usec (0d, 00:00:00)
	freq: 2412
	beacon interval: 100 TUs
	capability: ESS Privacy ShortSlotTime (0x0411)
	signal: -67.00 dBm
	last seen: 0 ms ago
[... 181 lines cut ...]
		 * PCO active: 0
		 * PCO phase: 0
	WMM:	 * Parameter version 1
		 * u-APSD
		 * BE: CW 15-1023, AIFSN 3
		 * BK: CW 15-1023, AIFSN 7
		 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
		 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec
	WPS:	 * Version: 1.0
		 * Wi-Fi Protected Setup State: 2 (Configured)
		 * Response Type: 3 (AP)
		 * UUID: 00000000-0000-1000-0000-e894f6535910
		 * Manufacturer: TP-LINK
		 * Model: TL-WR740N
		 * Model Number: 4.0
		 * Serial Number: 1.0
		 * Primary Device Type: 6-0050f204-1
		 * Device name: Wireless Router TL-WR740N
		 * Config methods: Ethernet, Label, PBC
		 * RF Bands: 0x1
		 * Unknown TLV (0x1049, 20 bytes): 00 24 e2 60 02 00 01 01 60 00 00 02 00 01 60 01 00 02 00 01

> As a workaround, note that the web interface accepts e.g. "https://confirm.html?ssid=MYSSID&encrypted=encrypted". Could you try that, to verify that authentication is possible?

Yes, it is possible.

davesteele, 2019-04-29:
I ran that output through the algorithm - All four APs listed as encrypted. I don't know where the problem is coming from.

davesteele, 2019-04-29:
Do you have more than one wifi adapter attached?

iv-med, 2019-04-29:
> Do you have more than one wifi adapter attached?

No, it is on the Raspberry Pi Zero W.

davesteele, 2019-04-29:
If you wait a while and then reload the page, does it fix itself?

danbachar, 2019-05-09:
+1 on the issue.

I'm running comitup on raspbian, some networks show up as unsecured even though they actually are.

APs: 
```
BSS f0:9f:c2:69:11:1a(on wlan0)
	TSF: 0 usec (0d, 00:00:00)
	freq: 5200
	beacon interval: 100 TUs
	capability: ESS Privacy SpectrumMgmt ShortSlotTime (0x0511)
	signal: -51.00 dBm
	last seen: 0 ms ago
	SSID: mantro
	Supported rates: 6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 
	DS Parameter set: channel 40
	BSS Load:
		 * station count: 12
		 * channel utilisation: 30/255
		 * available admission capacity: 0 [*32us]
	HT capabilities:
		Capabilities: 0x9ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-31
	HT operation:
		 * primary channel: 40
		 * secondary channel offset: below
		 * STA channel width: any
		 * RIFS: 0
		 * HT protection: no
		 * non-GF present: 1
		 * OBSS non-GF present: 0
		 * dual beacon: 0
		 * dual CTS protection: 0
		 * STBC beacon: 0
		 * L-SIG TXOP Prot: 0
		 * PCO active: 0
		 * PCO phase: 0
	Extended capabilities: Extended Channel Switching, TFS, WNM-Sleep Mode, TIM Broadcast, BSS Transition, SSID List, 6
	VHT capabilities:
		VHT Capabilities (0x338b79f2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			short GI (160/80+80 MHz)
			TX STBC
			SU Beamformer
			SU Beamformee
			MU Beamformer
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
[... 2331 lines cut ...]
		 * STA channel width: any
		 * RIFS: 1
		 * HT protection: no
		 * non-GF present: 1
		 * OBSS non-GF present: 0
		 * dual beacon: 0
		 * dual CTS protection: 0
		 * STBC beacon: 0
		 * L-SIG TXOP Prot: 0
		 * PCO active: 0
		 * PCO phase: 0
	Extended capabilities: 6
	WMM:	 * Parameter version 1
		 * BE: CW 15-1023, AIFSN 3
		 * BK: CW 15-1023, AIFSN 7
		 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
		 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec
	RSN:	 * Version: 1
		 * Group cipher: CCMP
		 * Pairwise ciphers: CCMP
		 * Authentication suites: PSK
		 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
```

An example of a network that is secured but shows up as unsecured is mantro.

davesteele, 2019-05-10:
@danbacher, that error message indicates that you are using Comitup 0.7 or earlier. That is really old. Try a newer version.

danbachar, 2019-05-10:
that's weird, as I installed comitup the day before through apt-get install. How should I install it otherwise?

davesteele, 2019-05-10:
Take a look at the [PPA page](https://davesteele.github.io/comitup/ppa.html), or [download](https://davesteele.github.io/comitup/archive.html) a more recent package.

davesteele, 2019-05-11:
@iv-med, what version of Comitup are you running?

danbachar, 2019-05-14:
I installed a newer version and most of my networks show up as unencrypted.

Same error message too:

[image]

Is there a command I could run to get comitup's version? @davesteele

davesteele, 2019-05-14:
That is not newer. You still have python2.7 and CherryPy in the output.

davesteele, 2019-05-14:
"dpkg --list comitup"

danbachar, 2019-05-15:
@davesteele fixed after upgrading to latest version

closed 2019-05-15 by iv-med