davesteele/comitup #39: After installing, WiFi does not work at all.

DaleMitchell, 2019-05-10:


davesteele, 2019-05-10:
I've probably never run setup.py on Raspbian. It serves as a config file for the debian build. Try the deb file.

davesteele, 2019-05-10:
It occurs to me that you may not have seen the [home page](https://davesteele.github.io/comitup/).

DaleMitchell, 2019-05-10:
Sorry if I came across as grouchy earlier, didn't get enough sleep.

I have seen the homepage, it sets high expectations for the software that I
like, but I'm frustrated that I'm getting these errors with the Python
files in the master repo! If I had a fix, I wouldn't hesitate to make a
pull request for the benefit of others.

davesteele, 2019-05-10:
You pretty much have to have a deb install for the environment to work. For development after that, I use a small deploy script to scp the project tree from a remote computer to a local project tree, and then copy the files to the individual subdirectories in /usr/share/comitup/. The last step could be skipped with some work to address the path problem you're seeing. But, I prefer to edit the project on a remote system, which means that some sort of custom deploy process is required for me.

DaleMitchell, 2019-05-10:
Okay... so the `setup.py` doesn't really do what people expect it to do?
There's more configuration that needs to be done beyond running `setup.py
build` and `setup.py install`? Why don't we have `setup.py` do everything
that needs to be done?

davesteele, 2019-05-10:
setup.py is intended to be cross platform, and currently does that job. The .deb file is specifically targeted to a particular OS, and can be uninstalled.

It sounds like you need to go find the tool that is right for you.

DaleMitchell, 2019-05-10:
Then why do I get:
```
ImportError: cannot import name persist
```

After running build and install with setup.py if it currently does the job?

davesteele, 2019-05-10:
Because there is a PYTHONPATH problem. You are welcome to attack that, if you wish. Or, you can vote with your feet.

closed 2019-05-11 by davesteele

DaleMitchell, 2019-05-11:
Why did you close an issue that isn't fixed? This issue doesn't have to do with a premade image.