davesteele/comitup #41: Startup time

krebbi, 2019-05-16:
Comitup needs a lot of time to start. Any ideas how I can speed it up? 

[image]

davesteele, 2019-05-16:
[here](https://github.com/davesteele/comitup/blob/ef4df2a96d6d53ced6ec39fa9e39f8aa21faa177/comitup/comitup.py#L98)

It gives time for the (sole) WiFi interface to get an initial list of APs.

krebbi, 2019-05-16:
can I reduce this to 2 or 3 ?

davesteele, 2019-05-16:
Sure. Keep an eye on the reported AP list.

krebbi, 2019-05-27:
Still have a lot of bootup-time thanks to comitiup. 
Do you think it's possible somehow to not block/delay the boot process this long?

davesteele, 2019-05-27:
Yes, but it is not a priority for me right now.

commit 60b53bf2 (2019-06-10): Add changelog note that startup time is improved

Closes #41
closed 2019-06-10 by davesteele