davesteele/comitup #44: Loss of networking, dual interface, state CONNECTED

davesteele, 2019-06-13:
With recent stretch updates, the PI is showing a loss of networking on all interfaces when CONNECTED in the dual wifi interface "router" mode. The outage includes the Ethernet interface.

The problem persists when any of the hotspot iptables rules, 'appliance' iptables rules, or WPA kick logic are disabled. 

All appears to work fine if:
  - there is only one interface,
  - or if there are two interfaces and appliance_mode is false

Note that the usb wifi interface configures to wlan0 on this particular system, and takes the role as hotspot.

Need to debug with KVM attached.

carlesgutierrez, 2019-06-14:
Dear Dave, thanks for all your software and efforts. This is really cool tool. 

I've been experiencing 2 similar issues... I do not fully understand what "dual wifi interface router mode" is. 

I'm using a MIFI " TP-Link Mobile Wi-Fi" configured as default (PDP Type: IPv4 & IPv6). Also I'm using my phone (Android) terminal as wifi router ( by default too ) that is giving better results. I've 2 rpi zero w with the lastest comitup installed from your Raspian Image and 2 more with comitup installed by terminal comands. 

My issues:
- Loss of network in just one of my four Raspberrypi Zero W (from the Raspian Image. This one is often lossing the network using MIFI or Android Terminal. (I've checked that have free space available)
- The other issue is about be able to get all 4 Raspberrys connected to MIFI or Android: MIFI only gives good result for one of them. Using my Android I can get all them connected at once but the loss of connectivity in that terminal remains.  

Is this about the same issue? Shall I try to modify my Router configuration?
Thanks!

davesteele, 2019-06-14:
It is what I call "appliance mode" in the conf file - two wifi interfaces, with routing and masquerading between the hotspot and the upstream connections.

I don't know details on what the problem is yet, or why ipv6 on a wifi interface would preclude ipv4 traffic on the etheret port. I just noticed that some interfaces are suddenly serving ipv6, and that the start of the problems coincided with the introductin of a new ISP-provided home router.

davesteele, 2019-06-17:
More news - the work computer (Debian Sid) connected to the Pi is, when it shares both and Ethernet and upstream "appliance mode" wifi connection with the Pi, appears to not respect its ipv4 routes when processing outbound packets destined for the Ethernet connection, and sends them out on the wifi interface instead (with a problematic source ip address for that interface).

davesteele, 2019-06-18:
Mitigated by adding a route on the desktop computer with a more restrictive netmask for the Ethernet connection, guaranteeing a win for that route. Not a Pi problem. Closing.

closed 2019-06-18 by davesteele