davesteele/comitup #45: Issue with Comitup Wifi-Manager installation

montup, 2019-06-20:
Hi,
To make pi 3 as an access point and wifi client I tried Comitup wifi manager.
First, I tried with Comitup image which is given in below link. and it was working properly.

https://davesteele.github.io/comitup/

After that, I tried with installing Comitup package with the official raspbian OS. I followed steps which are given in below link but it was not working for me.

https://davesteele.github.io/comitup/ppa.html

After booting I could 't see pi as an access point. Is there any other way to do this thing?
(there is no previous access point available)

Thanks!!

davesteele, 2019-06-20:
There is one extra step when installing on a Pi, as mentioned on the home page and the [Installing Comitup](https://github.com/davesteele/comitup/wiki/Installing-Comitup) wiki page.

montup, 2019-06-21:
Hi 
    Thanks so much!!
    Now it's working

closed 2019-06-21 by davesteele