davesteele/comitup #46: Hotspot on every reboot

kunalkanani93, 2019-06-29:
Hi Dave,

I am using Comitup from verion 1.4 and  updated to 1.5 now but I always encounter an issue that is on every reboot of RPI comitup hotspot shows up. Even when am in same network. 

I would like to know if I have not installed it properly or doing something wrong. Am just running a GUI app on startup with /etc/profile. 

Thanks

davesteele, 2019-06-30:
Are you using a Pi with built in wifi? Is there a second wifi adapter? Are you using the recent Buster Raspbian release?

kunalkanani93, 2019-07-01:
Yes am using built-in wifi, and no second wifi adapter is present. I am using previous release and now might go for Buster Raspbian release.

davesteele, 2019-07-01:
You can try debugging the connection via nmcli:

sudo systemctl stop comitup
nmcli dev disconnect wlan0
nmcli con up <SSID>

... and check in the logs for NetworkManager messages.

Try "iw dev wlan0 scan" to see if your AP is visible.

And don't be in a hurry to move to Buster.

kunalkanani93, 2019-07-02:
I did fresh installation now and everything seems ok, ill use this method if it happens again.

Right now, 
- after stopping comitup
- nmcli (showing wlan0 connected to my network)

OK will not move to Buster. Thank you

closed 2019-07-02 by kunalkanani93

kunalkanani93, 2019-07-25:
Hi,

How do I fix this error ? It happened suddenly and comitup is not up. I had cloned RPI image when everything was working fine but it also has same problem now.

[image]
[image]
[image]

davesteele, 2019-07-25:
See #48