davesteele/comitup #47: Modify comit-xx number. A clone Raspberrypi Issue.

carlesgutierrez, 2019-07-02:
Hello, 
 I've cloned a raspberrypi with comitup. So now their comitup name ( comit-xx) got exactly the same as the other one. I guess that was blocking one of them to allow conect to the network.  is that the right and expected behaviour, right?

A short try I did that works: remove the comitup package ( sudo apt-get remove comitup ) and install it again ( sudo apt-get install comitup ). The comitup name has changed ok. I've had 
 to add wifi password to this new one and now both are able to connect at same time. 

Is this a good process? Or may be all this was not necesary ;-D

krebbi, 2019-07-02:
So your idea is one should check if the chosen network name is already in the air and alter it if needed?

davesteele, 2019-07-02:
Yes, that is what I would expect to happen. In particular, avahi will refuse to publish a second "comitup-<nn>.local" on the network.

Note that the number is stored in /var/lib/comitup/comitup.json. Edit or delete that file and restart to change it.

I'm frankly surprised that we've encountered a collision already. If you are worried about it happening again, add an "n" or two to ap_name in comitup.conf.

carlesgutierrez, 2019-07-03:
Well I wanted to confirm if that behauviour was the espected. 

@krebbi that might be a solution, but then another issue might be to do not then be able to recognize who is each one. 

There is a way to force the number ( XX ) and the name of the network, isn't? Might be a better solution?

krebbi, 2019-07-03:
You can set a fix Wifi-Name for each device. 

See https://davesteele.github.io/comitup/man/comitup-conf.5.html

davesteele, 2021-03-09:
The number is stored in /var/lib/comitup.

closed 2021-03-15 by davesteele