davesteele/comitup #48: Captive Webseite doesn't connect to Wifi

krebbi, 2019-07-19:
Fresh install on Raspberry Pi Zero W. 

After connecting to the Hotspot the CaptivePortal shows up. I chose my network and enter the password (if required). The trying-to-connect-page show up und after a few seconds the list of available networks shows up again.

Connecting through comitup-cli to the same network works like a charm.

davesteele, 2019-07-20:
I just tried this with the shiny new v1.6-1. It worked fine on a Zero W, using the captive portal web page.

krebbi, 2019-07-21:
Where do I get the shiny new version? Is it available in your repo?

davesteele, 2019-07-22:
[Here](https://davesteele.github.io/comitup/archive.html), [here](https://davesteele.github.io/comitup/ppa.html), or [here](https://steele.debian.net/comitup/)

krebbi, 2019-07-22:
ohh, comitup-cli doesn't do anything anymore inthe shiny new 1.6-1.
[image]

davesteele, 2019-07-22:
Does the file /etc/NetworkManager/dnsmasq-shared.d/nm-dns-sabotage.conf exist? It should not anymore.

krebbi, 2019-07-23:
It does exist. I removed it, but that didn't solve the problem. 
comitup-cli still exits without any error-message.

davesteele, 2019-07-23:
I can't replicate. Perhaps a purge and reinstall?

krebbi, 2019-07-24:
sudo comitup fails:

`ERROR:dbus.connection:Unable to set arguments ({'ipv6': {'method': 'ignore'}, '802-11-wireless': {'mode': 'ap', 'band': 'bg', 'ssid': [dbus.Byte(99), dbus.Byte(111), dbus.Byte(109), dbus.Byte(105), dbus.Byte(116), dbus.Byte(117), dbus.Byte(112), dbus.Byte(45), dbus.Byte(55), dbus.Byte(49)]}, 'ipv4': {'method': 'manual', 'address-data': [{'prefix': 24, 'address': '10.42.0.1'}]}, 'connection': {'type': '802-11-wireless', 'id': 'comitup-71-0000', 'autoconnect': False, 'uuid': '04b6c41e-f159-4e59-b4bc-02e919a16e8a', 'interface-name': 'wlan0'}},) according to signature 'a{sa{sv}}': <class 'TypeError'>: an integer is required (got type str)
Traceback (most recent call last):
  File "/usr/sbin/comitup", line 11, in <module>
    load_entry_point('comitup==1.6', 'console_scripts', 'comitup-cmd')()
  File "/usr/share/comitup/comitup/comitup.py", line 114, in main
    cdns.state_callback,
  File "/usr/share/comitup/comitup/statemgr.py", line 171, in init_state_mgr
    conf.ap_password
  File "/usr/share/comitup/comitup/states.py", line 344, in init_states
    assure_hotspot(hotspot_name, modemgr.get_ap_device(), hotspot_pw)
  File "/usr/share/comitup/comitup/states.py", line 313, in assure_hotspot
    nm.make_hotspot(ssid, device, password)
  File "/usr/share/comitup/comitup/nm.py", line 290, in make_hotspot
    nm.Settings.AddConnection(settings)
  File "<string>", line 4, in AddConnection
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 641, in call_blocking
    message.append(signature=signature, *args)
TypeError: an integer is required (got type str)
`

davesteele, 2019-07-24:
That doesn't make sense. Are you on stretch?

krebbi, 2019-07-25:
Yes. 
Release 9.9
Kernel 4.19.57+

davesteele, 2019-07-25:
I've been using Buster. I'll try Stable this weekend.

reck7, 2019-07-25:
Today I ran updates on Pi Zero W and it picked up the 1.6-1 release and noticed an error on the install about issue starting the service. I appear to have this same issue. 

Honestly I am new to this project so don't have full understanding how it works. 

The one strange thing I have noticed is that like @krebbi I did get a TypeError of str rather than int.  But when I ran it again I get a TypeError: Expected a string or unicode object which contained my wifi psk. 

The only thing I have specifically modified was disable ipv6 by blacklisting it in /etc/modprobe.d/ipv6.conf, but this caused no issues with 1.5-1  I started with your stretch lite img when building this Pi

I have stopped the service, as it was looping trying to start every 15 sec. node-red is already a hog, didnt need things fighting :)

davesteele, 2019-07-25:
My current theory is that there is a bug in python-networkmanager, version 2.0.1-4, which does not handle NetworkManager configurations using the [newer address-data scheme](https://github.com/davesteele/comitup/commit/3d1adc614bee90a4a2f0bbaff849c98e2636c952#diff-cc65308170b7825e36a22d470fc2280fR267). A quick test of this would be to load [python-networkmanager version 2.1-1](https://archive.raspbian.org/raspbian/pool/main/p/python-networkmanager/) from Buster.

It will be a few days before I will be able to try this.

davesteele, 2019-07-25:
To be clear, the workarounds are to revert to 1.5-1 or to download a new (Buster) image.

reck7, 2019-07-25:
Thanks for looking into this. 

What is the proper way to revert to 1.5-1? I am not seeing it using the PPA.

pi@pi0w:~ $ sudo apt-cache madison comitup
   comitup |      1.6-1 | http://davesteele.github.io/comitup/repo comitup/main armhf Packages
   comitup |      0.7-1 | http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages

davesteele, 2019-07-25:
The archive is [here](https://davesteele.github.io/comitup/archive.html)

krebbi, 2019-07-25:
thx. as we are packaging your comitup with our software it's even better to use the deb-files and not the repo.

reck7, 2019-07-25:
Do you recommend going to python-networkmanager, version 2.0.1-3?

davesteele, 2019-07-25:
No.

krebbi, 2019-07-26:
Installing networkmanager 2.1-1 did the trick.

davesteele, 2019-07-26:
Thanks for checking this out. I've added the package to the comitup repository, and added a versioned dependency to comitup.

closed 2019-07-26 by davesteele

krebbi, 2019-07-29:
When I update the package from 1.4 to 1.6 the new networkmanager isn't installed automatically.
could you check within comitup to have networkmanager >=2.1*  and if not, try to install it?

kunalkanani93, 2019-07-29:
I just updated and its working now.  Thanks

davesteele, 2019-07-29:
82a6359f

commit 82a6359f (2019-07-26): Require a newer python3-networkmanager

Closes #48.