davesteele/comitup #49: DHCP server handing out dhcp offers while in state connected

reck7, 2019-07-26:
In playing around getting 1.6-1 running on raspberry pi zero I have found dhcp server continues to hand out 10.41.0.x addresses while device is in connects state.  I have a esp8266 that come out of deep sleep and connect to wifi. It appears that comitup is seeing the discover broadcasts and is sending out offers.  

[image]

This was in 1.5-1 as well.  Honestly I have not rebooted once got 1.6-1 running with the networkmanager 2.1-1. I am remote to where the device is and did not want to lose connectivity.

davesteele, 2019-07-26:
DHCP sticks around in dual-interface mode, but it shouldn't in the single interface case. I'll take a look at it.

davesteele, 2019-07-26:
dnsmasq management was moved from NetworkManager to Comitup in version 1.6.

reck7, 2019-07-26:
I did a purge of comitup, did an apt autoremove and the installed again to pick up your changes from today.  It still seems to be handing out DHCP offers. 
[image]

davesteele, 2019-07-26:
Today's changes addressed a different bug. 

The fix for this will look more like a0b2e2b66 (this is untested).

commit 0a09509a (2019-07-28): Disable dhcp in single interface connected mode

Closes #49
closed 2019-07-28 by davesteele