davesteele/comitup #52: Static IP configuration gets in the way with a network manager shared interface

muelli, 2019-07-30:
I am trying to bring comitup to a raspberry pi. In order to access the headless raspberry pi, I have a PC with two NICs. On that PC I have configured the second NIC through network manager to share the Internet connection of the other NIC. Network manager seems to use IP addresses in the range of `10.42.0.1/32` for the clients connecting through the second interface.  This is the exact range that comitup uses :(  I think I experience problems due the IP network ranges being the same.

Would it be possible to detect whether the IP addresses comitup wants to hand out to its clients clashes with a subnet the device is already part of?
As a stop-gap measure, using a different range than the one network manager uses by default might be okay.

FWIW: I have been able to convince network manager to hand out different IPs: https://askubuntu.com/a/829759/207933 but still think that it'd be valuable if comitup did not hand out IPs that it can't handle.

davesteele, 2019-07-30:
I'm not sure I follow. Are we talking about #49?

If not, 10.42.0.0/24 is also [hard coded](https://github.com/davesteele/comitup/blob/b497c1ff06b18c8b34c7fd1e9e3a1125b4a164aa/comitup/nm.py#L271) into Comitup (copied from NM). It could be changed, made into a config parameter.

muelli, 2019-07-30:
> I'm not sure I follow. Are we talking about #49?
> 
sorry  for not being more precise :-/ 

> If not, 10.42.0.0/24 is also [hard coded](https://github.com/davesteele/comitup/blob/b497c1ff06b18c8b34c7fd1e9e3a1125b4a164aa/comitup/nm.py#L271) into Comitup (copied from NM). It could be changed, made into a config parameter.

yeah, I'm talking about the hardcoded information which I suspect causes a conflict when plugging the ethernet cable of the Raspberry Pi (with Comitup running) into a port on a PC using network-manager to share an Internet connection.
I think Comitup cannot work as expected if the Raspberry Pi's wired network interface has an IP address in the same subnet which is used for connecting to wireless clients.
So using a working subnet would be best.

closed 2019-07-30 by davesteele

muelli, 2019-07-30:
cool. thanks.

This works for as long as there is no interface with that subnet. The odds are small, but it might happen.

commit 25e34f92 (2019-07-30): Move the hotspot net from 10.42.0.0/24 to 41

Closes #52

Remove a conflict with a host which is also sharing a hotspot
created by NetworkManager.