davesteele/comitup #54: Question: Is it possible to stop hotspot when an edge device successfully connects a Network?

Sharvin26, 2019-08-01:
Hello

I have a Raspberry Pi ( This is the edge device where I have configured the comitup service ) with the Raspbian ( Stretch ).

## Expected Behavior

I have a requirement as follows =>

- I want to connect an edge device to the available network. 

- The access point on the edge device will only stop when the edge device is successfully connected to the wireless network. 

**Reason** behind this requirement is that when a wrong password is sent or due to any circumstance it is not possible to connect that wifi then, in that case, I won't get feedback if the edge device is connected successfully or not.

## Actual Behavior

I saw a behavior whenever a connect API is called on the edge device it stops the Hotspot ( i.e. It's own AP ) and tries to connect the wireless network / AP

You have also mentioned this in the documentation

> If the computer cannot automatically connect to a local wifi access point, comitup will create a custom hotspot, and establish a comitup-web web service on that network. 

## Steps to Reproduce the Problem
1.  Added the line `deb http://davesteele.github.io/comitup/repo comitup main
` inside this file =>` /etc/apt/sources.list`.
2. `wget https://davesteele.github.io/key-366150CE.pub.txt`
3. `sudo apt-key add key-366150CE.pub.txt`
4. `sudo apt-get update`
5. `sudo apt-get install comitup`
6. I connected the Hotspot that was created by the edge device
7. After that, I went to the `<edge_device_ip_address>:80` on the browser and I got all the available network list. 
8. I selected the network and typed in the wrong password and pressed the connect.
9. In that case, it turned off the Access point and tried to connect the Wireless network.
10. After a few seconds of try it started it's Hotspot again.

## Question

So I was wondering if is it possible to stop the Hotspot only if the Edge device is successfully connected to the wifi?

davesteele, 2019-08-01:
There is currently no mode to do exactly what you want. Comitup has to drop the AP in order to use the interface to attempt the upstream connection. Failure is indicated by the recurrence of the Comitup AP.

There is a dual interface mode that preserves the Comitup AP, but it does not do significantly more to report connection errors.

Sharvin26, 2019-08-02:
Thanks for the response @davesteele 
How can I enable the dual interface mode?

davesteele, 2019-08-02:
Add a USB Wifi interface. See also *enable_appliance_mode* and *primary_wifi_device* in the [configuration file](https://davesteele.github.io/comitup/man/comitup-conf.pdf).

closed 2019-08-02 by Sharvin26