davesteele/comitup #56: web service throws exception

muelli, 2019-08-03:
I have just installed comitup from the PPA on my fresh Raspian installation.

I have stopped systemd-resolved (https://github.com/davesteele/comitup/issues/51) and restarted comitup. It spawned the wifi, I connected and used a Browser to connect to the IP address of the device (10.42.0.1).

I expected a Web interface, but I got this exception:

```
dbus.exceptions.DBusException

dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.12 was not provided by any .service files

Traceback (most recent call last)

File "/usr/lib/python3/dist-packages/flask/app.py", line 2309, in __call__

return self.wsgi_app(environ, start_response)

File "/usr/lib/python3/dist-packages/flask/app.py", line 2295, in wsgi_app

response = self.handle_exception(e)

File "/usr/lib/python3/dist-packages/flask/app.py", line 1741, in handle_exception

reraise(exc_type, exc_value, tb)

File "/usr/lib/python3/dist-packages/flask/_compat.py", line 35, in reraise

raise value

File "/usr/lib/python3/dist-packages/flask/app.py", line 2292, in wsgi_app

response = self.full_dispatch_request()

File "/usr/lib/python3/dist-packages/flask/app.py", line 1815, in full_dispatch_request

rv = self.handle_user_exception(e)

File "/usr/lib/python3/dist-packages/flask/app.py", line 1718, in handle_user_exception

reraise(exc_type, exc_value, tb)

File "/usr/lib/python3/dist-packages/flask/_compat.py", line 35, in reraise

raise value

File "/usr/lib/python3/dist-packages/flask/app.py", line 1813, in full_dispatch_request

rv = self.dispatch_request()

File "/usr/lib/python3/dist-packages/flask/app.py", line 1799, in dispatch_request

return self.view_functions[rule.endpoint](**req.view_args)

File "/usr/share/comitup/web/comitupweb.py", line 61, in index

points = ciu_client.ciu_points()

File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__

**keywords)

File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking

message, timeout)

dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.12 was not provided by any .service files

The debugger caught an exception in your WSGI application. You can now look at the traceback which led to the error.

To switch between the interactive traceback and the plaintext one, you can click on the "Traceback" headline. From the text traceback you can also create a paste of it. For code execution mouse-over the frame you want to debug and click on the console icon on the right side.

You can execute arbitrary Python code in the stack frames and there are some extra helpers available for introspection:

dump() shows all variables in the frame

dump(obj) dumps all that's known about the object

Brought to you by DON'T PANIC, your friendly Werkzeug powered traceback interpreter.
```

Note that I haven't don't anything else but installing the package, most notably, I haven't changed the configuration.

davesteele, 2019-08-05:
You will probably need to debug from an Ethernet connection. If you downloaded a comitup image, it is at 10.0.0.2/24. 

Is comitup running? 

    systemctl status comitup

If not, what is the trace?

    sudo comitup

davesteele, 2019-08-09:
There are a number of fixes in v1.7 which could explain this. I'm closing this until such time that it is reported for that version.

muelli, 2019-08-09:
It's still happening.

```
$ comitup --version
Comitup requires root privileges
$ sudo comitup --version
usage: comitup [-h] [-c]
comitup: error: unrecognized arguments: --version
$ apt-cache policy comitup
comitup:
  Installed: 1.7-1
  Candidate: 1.7-1
  Version table:
 *** 1.7-1 500
        500 http://davesteele.github.io/comitup/repo comitup/main armhf Packages
        100 /var/lib/dpkg/status
     1.3.1-1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages

$ systemctl status comitup
● comitup.service - Comitup Wifi Management
   Loaded: loaded (/lib/systemd/system/comitup.service; enabled; vendor preset: 
   Active: active (running) since Fri 2019-08-09 07:24:10 UTC; 8min ago
     Docs: man:comitup(8)
 Main PID: 1659 (comitup)
    Tasks: 2 (limit: 2319)
   Memory: 12.0M
   CGroup: /system.slice/comitup.service
           ├─1659 /usr/bin/python3 /usr/sbin/comitup
           └─1784 dnsmasq --conf-file=/usr/share/comitup/dns/dns-hotspot.conf --
```

Once I remove the systemd-resolved listener, i.e. by configuring DNSStubListener=no in /etc/systemd/resolved.conf, it works much better.
So maybe this can be closed as a duplicate of https://github.com/davesteele/comitup/issues/51, although it should probably be either more robust or be more informative in its logging.

davesteele, 2019-08-09:
Reopening for logging improvement.

commit 3f466a88 (2021-10-18): Add comitupweb logging

Closes #56