davesteele/comitup #58: wireless failed after upgrade

bretonium, 2019-08-10:
I have booted my rpi3b from `2019-07-20-Comitup.img`. Probably after `apt-get upgrade` wireless fails to go up. comitup cannot create a hotspot and nm cannot connect to the configured ap. It leads to many tracebacks like this:

```
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/sbin/comitup", line 11, in <module>
Aug 10 18:38:17 raspberrypi comitup[1039]:     load_entry_point('comitup==1.7', 'console_scripts', 'comitup-cmd')()
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/comitup.py", line 114, in main
Aug 10 18:38:17 raspberrypi comitup[1039]:     cdns.state_callback,
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/statemgr.py", line 171, in init_state_mgr
Aug 10 18:38:17 raspberrypi comitup[1039]:     conf.ap_password
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/states.py", line 347, in init_states
Aug 10 18:38:17 raspberrypi comitup[1039]:     set_state('HOTSPOT')
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/states.py", line 280, in set_state
Aug 10 18:38:17 raspberrypi comitup[1039]:     state_info.start_fn()
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/states.py", line 56, in wrapper
Aug 10 18:38:17 raspberrypi comitup[1039]:     returnvalue = fn(*args, **kwargs)
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/states.py", line 109, in hotspot_start
Aug 10 18:38:17 raspberrypi comitup[1039]:     activate_connection(hs_ssid, 'HOTSPOT')
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/states.py", line 296, in activate_connection
Aug 10 18:38:17 raspberrypi comitup[1039]:     path=path)
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/share/comitup/comitup/nm.py", line 178, in activate_connection_by_ssid
Aug 10 18:38:17 raspberrypi comitup[1039]:     nm.NetworkManager.ActivateConnection(connection, device, path)
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "<string>", line 6, in ActivateConnection
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
Aug 10 18:38:17 raspberrypi comitup[1039]:     **keywords)
Aug 10 18:38:17 raspberrypi comitup[1039]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
Aug 10 18:38:17 raspberrypi comitup[1039]:     message, timeout)
Aug 10 18:38:17 raspberrypi comitup[1039]: dbus.exceptions.DBusException: org.freedesktop.NetworkManager.UnknownConnection: Connection 'comitup-42-0000' is not available on device wlan0 because device is not available
Aug 10 18:38:17 raspberrypi systemd[1]: comitup.service: Main process exited, code=exited, status=1/FAILURE
```

I am not sure that it happened after upgrade, but my impression so far is that upgrade caused that. Other than upgrade, i removed mentions of eth0 from `/etc/network/interfaces`. The issue happened twice.

Investigation showed that issue #12 happened, and removing `wpa_supplicant.conf` helped.

bretonium, 2019-08-10:
after some reading, i think the user from https://github.com/davesteele/comitup/issues/31#issuecomment-472021820 is having the same problem

davesteele, 2019-08-11:
It is wlan0 (actually, any wlan which is to be managed by Comitup) which needs to be removed from /etc/interfaces. The Comitup images add an entry to the file for a convenience static configuration for eth0.

*Connection 'comitup-42-0000' is not available on device wlan0 because device is not available*

This suggests that wlan0 has an entry in that file. But, that should not happen if you are using a Comitup image.

There is another recent problem caused by an obsolete configuration file under NetworkManager. That should only be in play if you started with a version from before 2019-07-20.

bretonium, 2019-08-11:
> It is wlan0 (actually, any wlan which is to be managed by Comitup) which needs to be removed from /etc/interfaces. The Comitup images add an entry to the file for a convenience static configuration for eth0.

There was nothing about `wlan0` in the `interfaces` files, there was only `eth0`. I wanted to let networkmanager manage `eth0` and had to remove the entries. Btw it would be good to actually remove those lines by default and let dhcp do its magic, but that is a story for another ticket.

> This suggests that wlan0 has an entry in that file. But, that should not happen if you are using a Comitup image.

Nope, `wlan0` was managed by `wpa_supplicant` and removing `wpa_supplicant.conf` helped to get `wlan0` under control of nm.

davesteele, 2019-08-11:
> There was nothing about `wlan0` in the `interfaces` files, there was only `eth0`. I wanted to let networkmanager manage `eth0` and had to remove the entries. Btw it would be good to actually remove [...]

The goal was to have two aliased IPs on eth0 - one static and one provided by DHCP. I haven't gotten that to work.

davesteele, 2019-10-02:
The wpa_supplicant contention is covered in the [wiki](https://github.com/davesteele/comitup/wiki/Installing-Comitup). Closing

closed 2019-10-02 by davesteele