davesteele/comitup #59: Unstable wifi

krebbi, 2019-08-15:
Hi Dave,

I have some devices in an environment where the sifi connection is not really reliable. When the connection drops, the Raspi is going to Hotspot mode. Is there any chance to let nm retry / reconnect to known networks?

And as secondary question: Is there a way to donate/pay/send something to you for all your efforts?

muelli, 2019-08-15:
hm. I was thinking whether it'd be interesting to have comitup separate those two steps, i.e. the setup phase and the bridging phase.
Then, it could spawn an "setup upstream wifi" hotspot for setting up, well, the upstream wifi connection and once that has completed, spawn a differently named hotspot, say "bridged Internet", to connect to the Internet. And you could configure when you'd want to go back to setup mode.

davesteele, 2019-08-16:
@krebbi, it will attempt to connect to its list of candidates in that condition, but only after a 3 minute timeout. That could probably be easily shortened when coming from the connected_fail case.

I haven't looked into funding options. I just added myself to the GitHub Sponsor program wait list. That's likely more trouble than it is worth.

@muelli, you can approximate what you are asking for by adding a USB wifi adapter. There's no web interface once connected, but the Comitup hotspot remains, and comitup-cli can be accessed via SSH.

Just don't use an adapter with the RTL8188CUS chipset. On Buster-era Raspbian, connection requests to it result in an overall kernel panic. It's unlikely that Comitup is part of the problem, but I've only found [one other](https://www.raspberrypi.org/forums/viewtopic.php?t=245127) report.

krebbi, 2019-08-16:
do you see any problems reducing this timeout to 30s ? and does it retry every X seconds endless or does it stop sometime?

davesteele, 2019-08-16:
At 30 secs, connection attempts get in the way of hotspot availability. I can add a one-time 5 sec timeout when dropping out of connected mode.

krebbi, 2019-09-02:
Could you please add the ability to configure this behaviour? Let's say: try to connect every X seconds to known networks and a try-to-reconnect-on-dropping-network thing?

davesteele, 2019-09-02:
Reconnect-on-dropping is in git. See the commit above.

davesteele, 2019-10-02:
A work-around is in git. Closing.

closed 2019-10-02 by davesteele