davesteele/comitup #62: Not opening wifi sign in page in mobile phones

krishnakulkarni88, 2019-09-27:
Hi,

It's a great project, It was very helpful for my college project. when pi is in access point mode for connecting wifi I am not able to open the sign-in page(comitup-94.local/) on my mobile phone. I have checked in other mobiles also but it was not opening. If I open in laptop it was happening and able to connect with other wifi networks can I know any fix for this issue.

muelli, 2019-09-30:
I can only speculate.
Do you have any other network connection configured, e.g. mobile data?
It /could/ be that your operating system decides to not resolve the name via the WiFi, because it detected that there is no Internet connection. It'd be terribly incompatible with how captive portals are deployed today, though. Anyway, have you tried navigating to the IP address of the comitup device instead?

davesteele, 2019-10-01:
I just tried the latest with my Android phone. After connecting to the "comitup-nn" hotspot:

  * The captive portal detection worked - the AP selection page appeared unprompted.
  * comitup-nn.local and http://comitup-nn did not resolve correctly. Note that support for mdns/ZeroConf is not universal.
  * http://10.41.0.1 (note, this was 10.42.0.1 in older versions) works.

I'm guessing your laptop is a Mac.

davesteele, 2019-11-25:
In summary, automated captive portal detection should work with all-but-very-old releases, but mdns resolves will probably [never happen](https://stackoverflow.com/questions/30449988/how-to-enable-mdns-support-in-android-browser-address-bar)

closed 2019-11-25 by davesteele