davesteele/comitup #64: comitup doesn't start if avahi daemon isn't installed

Baltix, 2019-10-03:
comitup doesn't start if avahi daemon isn't installed - I get these error messages in /var/log/daemon.log

Sep 26 10:40:14 Baltix-Linux-PI systemd[1]: Starting Comitup Wifi Management...
Sep 26 10:40:15 Baltix-Linux-PI systemd[1]: Started Comitup Wifi Management.
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]: Traceback (most recent call last):
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     return self.get_name_owner(bus_name)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     's', (bus_name,), **keywords)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     message, timeout)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.freedesktop.Avahi': no such name
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]: During handling of the above exception, another exception occurred:
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]: Traceback (most recent call last):
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/sbin/comitup", line 11, in <module>
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     load_entry_point('comitup==1.3', 'console_scripts', 'comitup-cmd')()
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/share/comitup/comitup/comitup.py", line 104, in main
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     [webmgr.state_callback, iptmgr.state_callback],
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/share/comitup/comitup/statemgr.py", line 160, in init_state_mgr
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     states.set_state('HOTSPOT', timeout=5)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/share/comitup/comitup/states.py", line 287, in set_state
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     state_info.start_fn()
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/share/comitup/comitup/states.py", line 54, in wrapper
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     returnvalue = fn(*args, **kwargs)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/share/comitup/comitup/states.py", line 106, in hotspot_start
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     mdns.clear_entries()
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/share/comitup/comitup/mdns.py", line 113, in clear_entries
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     establish_group()
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/share/comitup/comitup/mdns.py", line 45, in establish_group
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     bus.get_object(DBUS_NAME, DBUS_PATH_SERVER),
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     follow_name_owner_changes=follow_name_owner_changes)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     self._named_service = conn.activate_name_owner(bus_name)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     self.start_service_by_name(bus_name)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     'su', (bus_name, flags)))
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]:     message, timeout)
Sep 26 10:40:16 Baltix-Linux-PI comitup[793]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Avahi was not provided by any .service files
Sep 26 10:40:16 Baltix-Linux-PI systemd[1]: comitup.service: Main process exited, code=exited, status=1/FAILURE
Sep 26 10:40:16 Baltix-Linux-PI systemd[1]: comitup.service: Failed with result 'exit-code'.

This issue can be fixed by using try ... except Exception , I've noticed, that other open source projects uses this approach, see for example https://bugs.gentoo.org/532328 or several examples at https://programcreek.com/python/example/5282/dbus.DBusException

I'm attaching a simple patch which solves startup issue, but I don't know if comitup works correctly without avahi daemon, it would be nice if comitup could work correctly without avahi daemon installed.
[mdns.py-fix-no-avahi-daemon.patch.txt](https://github.com/davesteele/comitup/files/3685177/mdns.py-fix-no-avahi-daemon.patch.txt)

closed 2019-10-03 by davesteele

Baltix, 2019-10-03:
So, why comitup can't work correctly without avahi daemon installed? Could you improve comitup to have an ability without avahi daemon?

davesteele, 2019-10-03:
Comitup's mission is to set up networking on a headless, wireless system. In that environment, how do you determine the IP address of the device once it is hooked up to the upstream AP? Avahi publishes that information n an easily-discoverable way.

I would not consider the ability to work without that capability to be an improvement.

commit 07c40f78 (2019-10-03): Add a Depends on Avahi

The daemon is a key component for finding the ip address for the
upstream AP connection.

Closes #64