davesteele/comitup #65: How to connect to WPA2 Enterprise network

bkrajendra, 2019-10-15:
is there any way to connect to WPA2 Enterprise network.

davesteele, 2019-10-15:
I've never had access to such a network to develop/test against.

bkrajendra, 2019-10-16:
Almost every industry uses WPA2 Enterprise for their WiFi network, which make all of existing IoT modules useless in such settings.
Ive tried ESP8266, ESP32 and now Raspberry Pi but all are useless.
Spent a lot of time figuring out to do it in some way but no success yet.

krebbi, 2019-10-16:
See 3.14 in the Docs for ESP8266 Support

https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf

davesteele, 2019-10-16:
Try this:

  - Attempt a connection using the comitup-web service or with comitup-cli. Does it show as a password-protected connection?
  - Edit the NetworkManager settings file for the connection (in /etc/NetworkManager/system-connections) to work with WPA2-Enterprise (guidance e.g. https://github.com/wylermr/NetworkManager-WPA2-Enterprise-Setup).
  - Verify the config using "nmcli con up <connection_name>"
  - Send me info on your operational config. Obscure the password.
  - Bonus points if you can get the output of the script "/usr
/share/comitup/comitup/nm.py listaccess". I need to see the "WpaFlags" returned by NetworkManager.

bkrajendra, 2019-10-16:
@krebbi  It did  not work for me.

bkrajendra, 2019-10-16:
@davesteele thanks a lot for help, I'll definitely go through all and report you back.

CallumJHays, 2019-11-20:
Hey guys, unsure if you managed to fix this. Turns out this is due to a bug being shipped since mid-Jessie: [bug forum discussion](https://www.raspberrypi.org/forums/viewtopic.php?t=247310)

I want to make it easier for members of my university club to experiment with Raspberry Pi's, and this repo is just what I was looking for!

Keen to add WPA2-Enterprise support (with both username, password, and perhaps optional ca-certificate) to the web-page. Will take a look at accomplishing that over the next few days and hopefully submit a PR.

CallumJHays, 2019-11-20:
For reference the solution was to edit:
`/etc/dhcpcd.conf`

and ensure that `wext` comes before `nl80211`

```
# Replace with your actual wireless interface name.
interface wlx******
env ifwireless=1
env wpa_supplicant_driver=wext,nl80211
```

closed 2021-04-10 by davesteele