davesteele/comitup #66: Hide Access point(SSID)

Muzzamil1, 2019-10-22:
Hi, @davesteele  Thanks for the Great project!
I just want to ask, is it possible to hide the access point(SSID)?

davesteele, 2019-10-22:
Yiou can make the AP hidden by adding "hidden=TRUE" to the "[wifi]" section of the NetworkManager configuration file in /etc/NetworkManager/system-connections. (See https://developer.gnome.org/NetworkManager/stable/nm-settings.html)

Login authentication for upstream APs is discussed in #65.

Muzzamil1, 2019-10-22:
Ok, thanks. I will try.

Muzzamil1, 2019-10-23:
Hi, @davesteele  I try to hide the network but it didn't work. 
I did the following steps:
`cd /etc/NetworkManager/system-connections`
`sudo nano wifiTest-0000.nmconnection`
and added `hidden=TRUE` in wifi section.

[image]

Then I clear the previously saved configuration using  `comitup-cli` and then reboot the Pi.
But after booting I can **still see the Access Point**. Then I go to `wifiTest-0000.nmconnection` in which I added the `hidden=TRUE` and that **line** was **no** longer there.
Am I missing something or doing something wrong?

davesteele, 2019-10-23:
Sorry about that. I remembered last night that the Hotspot config is currently remade on every connect. Currently, you would need to add this config item to the [source code](https://github.com/davesteele/comitup/blob/4694554df8e826f7e024f4cfd0172123efed60b3/comitup/nm.py#L262).

Muzzamil1, 2019-10-24:
I added `'hidden': True,` , then clear the previously saved configuration using `comitup-cli` and then reboot the Pi but I can still see the **AP** on my mobile.

[image]

davesteele, 2019-10-24:
You are charting new territory at this point. Check if the config file matches what you expect.

Muzzamil1, 2019-10-24:
I checked 
`sudo nano /etc/NetworkManager/system-connections/wifiTest-0000.nmconnection`
But there is **no hidden=True** in that file.

davesteele, 2019-10-24:
Ok, so check the documentation for NetworkManager AddConnection for the right parameter.

closed 2020-06-17 by davesteele