davesteele/comitup #70: DNS name

santhoshnp, 2019-11-08:
Hi team,

How to change the dns name instead 10.41.0.1 can you please help me?

davesteele, 2019-11-08:
The name can be set in the [configuration file](https://davesteele.github.io/comitup/man/comitup-conf.5.html). The ip address is [hard coded](https://github.com/davesteele/comitup/blob/master/comitup/nm.py#L271).

santhoshnp, 2020-01-10:
hi team,
I tried to change the DNS name but i am not getting the result can say which location you mentioning in the configuration file. why because i have done wrong first show again i have rebooted. i am new to this

santhoshnp, 2020-02-21:
I also tried but I am unable to change the DNS name in configuration file can you say which place it is

davesteele, 2020-02-21:
Note that, in connected mode, the "comitup-&lt;nn&gt;.local" "DNS name" is actually published using mdns, so your client must support ZeroConf for it to work. That support is [not universal](https://github.com/davesteele/comitup/wiki/ZeroconfBrowser).

santhoshnp, 2020-05-02:
Hi, davesteele,
Finally, I got the solution.Thank you so much for the reply

santhoshnp, 2020-05-02:
I also have another doubt how to add ssh in the website.

closed 2020-06-17 by davesteele