davesteele/comitup #71: The way to connect to Wi-Fi on iPhone

ticat-dev, 2019-11-09:
Hi daves, Thanks for your great project!
I'm testing the comitup now and I have one question for the way to connect to Wifi on iPhone.

When I chose and connected comitup-29 in my iPhone setting display, my iPhone wifi icon would never come out and then the captive portal come out without the wifi icon as below.
As a result my RP didn't get IP address.
[image]

And then I tried to push a cancelation button on the captive portal.
I got comitup AP after I pushed "Use Without Internet" button.
However the captive portal had never come out.

I used "http://10.41.0.1" on the safari browser because I have no choice without it.
And I chose my SSID and got the connectivity from my RP as below.
[image]

Is the process to give my RP the connectivity like above correct way?
It looks strange for captive portal cancelation process.
Please let me know the correct way to use the comitup if I did some wrong way.

Just so you know that I'm using iPhone XR and Raspbian GNU/Linux 10 (buster) for my RP.

davesteele, 2019-11-09:
That''s a bug. 

From an iPad:

    Nov  9 04:54:18 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:18] "GET /hotspot-detect.html HTTP/1.0" 302 -
    Nov  9 04:54:20 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:20] "GET / HTTP/1.0" 200 -
    Nov  9 04:54:21 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:21] "GET /hotspot-detect.html HTTP/1.1" 302 -
    Nov  9 04:54:21 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:21] "GET / HTTP/1.1" 200 -
    Nov  9 04:54:22 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:22] "GET /css/uikit.css HTTP/1.1" 200 -
    Nov  9 04:54:22 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:22] "GET /js/uikit-icons.min.js HTTP/1.1" 200 -
    Nov  9 04:54:22 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:22] "GET /js/uikit.js HTTP/1.1" 404 -
    Nov  9 04:54:22 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:22] "GET /hotspot-detect.html HTTP/1.0" 302 -
    Nov  9 04:54:22 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:22] "GET / HTTP/1.0" 200 -
    Nov  9 04:54:28 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:28] "GET /confirm?ssid=Fioptics01678&encrypted=encrypted HTTP/1.1" 200 -
    Nov  9 04:54:28 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:28] "GET /js/uikit.js HTTP/1.1" 404 -
    Nov  9 04:54:28 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:28] "GET /css/uikit.css HTTP/1.1" 304 -
    Nov  9 04:54:28 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:28] "GET /js/uikit-icons.min.js HTTP/1.1" 304 -
    Nov  9 04:54:28 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:28] "GET /hotspot-detect.html HTTP/1.0" 302 -
    Nov  9 04:54:30 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:30] "GET / HTTP/1.0" 200 -
    Nov  9 04:54:44 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:44] "POST /connect HTTP/1.1" 200 -
    Nov  9 04:54:44 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:44] "GET /js/uikit.js HTTP/1.1" 404 -
    Nov  9 04:54:44 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:44] "GET /css/uikit.css HTTP/1.1" 304 -
    Nov  9 04:54:44 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:44] "GET /js/uikit-icons.min.js HTTP/1.1" 304 -
    Nov  9 04:54:45 raspberrypi comitup-web[1072]: 10.41.0.134 - - [09/Nov/2019 04:54:45] "GET /hotspot-detect.html HTTP/1.0" 302 -
    Nov  9 04:54:45 raspberrypi comitup-web[1072]: Process Process-1:
    Nov  9 04:54:45 raspberrypi comitup-web[1072]: Traceback (most recent call last):
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:     self.run()
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:     self._target(*self._args, **self._kwargs)
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:   File "/usr/share/comitup/web/comitupweb.py", line 53, in do_connect
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:     ciu_client.ciu_connect(ssid, password)
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:   File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:     **keywords)
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    Nov  9 04:54:45 raspberrypi comitup-web[1072]:     message, timeout)
    Nov  9 04:54:45 raspberrypi comitup-web[1072]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.AccessDenied: Message has the same reply serial as a currently-outstanding existing method call

Connects work from Android, but returns:

    Nov  9 05:01:36 raspberrypi comitup-web[1072]: 10.41.0.141 - - [09/Nov/2019 05:01:36] "POST /connect HTTP/1.1" 200 -
    Nov  9 05:01:36 raspberrypi comitup-web[1072]: 10.41.0.141 - - [09/Nov/2019 05:01:36] "GET /js/uikit.js HTTP/1.1" 404 -
    Nov  9 05:01:36 raspberrypi comitup-web[1072]: 10.41.0.141 - - [09/Nov/2019 05:01:36] "GET /css/uikit.css HTTP/1.1" 304 -
    Nov  9 05:01:36 raspberrypi comitup-web[1072]: 10.41.0.141 - - [09/Nov/2019 05:01:36] "GET /js/uikit-icons.min.js HTTP/1.1" 304 -
    Nov  9 05:01:36 raspberrypi comitup-web[1072]: 10.41.0.141 - - [09/Nov/2019 05:01:36] "GET /img/favicon.ico HTTP/1.1" 500 -
    Nov  9 05:01:37 raspberrypi comitup-web[1072]: Traceback (most recent call last):
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 2309, in __call__
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:     return self.wsgi_app(environ, start_response)
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 2295, in wsgi_app
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:     response = self.handle_exception(e)
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1741, in handle_exception
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:     reraise(exc_type, exc_value, tb)
[... cut ...]
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1813, in full_dispatch_request
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:     rv = self.dispatch_request()
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1799, in dispatch_request
    Nov  9 05:01:37 raspberrypi comitup-web[1072]:     return self.view_functions[rule.endpoint](**req.view_args)
    Nov  9 05:01:37 raspberrypi comitup-web[1072]: TypeError: favicon() missing 1 required positional argument: 'path'

ticat-dev, 2019-11-25:
Hi Dave,

Thanks for your fixing this bug.

I did the apt-get update and check the programs change as ([8946c71](https://github.com/davesteele/comitup/commit/8946c71eeb0552dc5f2eb7fc3bcdfc1feb146020))
directly.
I checked that all changes was changed completely and tried same thing several times.

However it seemed the same as before, my wifi icon didn't appear with the captive portal at the same time even once.
Of course If I pushed "Use Without Internet" button and used "http://10.41.0.1" with safari, I can use the comitup was available.

Could you let me know if there are any other conceivable cause at the moment?
I want to try it if any.

davesteele, 2019-11-25:
The logs I listed above were from /var/log/comitup-web.log. Do you have any tracebacks in that file?

I was using a late-model iPad.

ticat-dev, 2019-12-06:
Hi Dave,

Sorry for my late reply and thanks for your updating information.
In fact I couldn't login to my RP for a router problem.
I bought a new router, set it, did "apt-get update" and tried connect to comitup today.
Also I checked that the current ([8946c71](https://github.com/davesteele/comitup/commit/8946c71eeb0552dc5f2eb7fc3bcdfc1feb146020)) was reflected.

However the result was same as previous process when I pushed blue-## on my iPhone.
The log in /var/log/comitup-web.log is as below and it doesn't look strange...

`2019-12-06 22:09:36,171 - comitup_web - INFO - Starting comitup-web
2019-12-06 22:11:10,645 - comitup_web - INFO - index.html - 9 points
2019-12-06 22:11:11,467 - comitup_web - INFO - index.html - 9 points
2019-12-06 22:11:12,058 - comitup_web - INFO - index.html - 9 points
2019-12-06 22:11:21,938 - comitup_web - INFO - index.html - 11 points
2019-12-06 22:11:22,466 - comitup_web - INFO - index.html - 11 points
2019-12-06 22:11:22,941 - comitup_web - INFO - index.html - 11 points
2019-12-06 22:11:28,064 - comitup_web - INFO - confirm.html - ssid xxxx, mode single
2019-12-06 22:11:29,518 - comitup_web - INFO - index.html - 12 points
2019-12-06 22:11:50,222 - comitup_web - INFO - index.html - 12 points
2019-12-06 22:11:50,625 - comitup_web - INFO - index.html - 12 points
2019-12-06 22:11:50,740 - comitup_web - INFO - index.html - 12 points
2019-12-06 22:11:50,835 - comitup_web - INFO - index.html - 12 points
2019-12-06 22:11:50,963 - comitup_web - INFO - index.html - 12 points
2019-12-06 22:11:51,532 - comitup_web - INFO - confirm.html - ssid xxxx, mode single
2019-12-06 22:11:59,289 - comitup_web - INFO - connect.html - ssid xxxx
2019-12-06 22:14:31,146 - comitup_web - INFO - Starting comitup-web`

jerbzz, 2021-03-23:
Hello @davesteele. I'm experiencing an issue on my iPhone 8 Plus (iOS 14.4.1).

I've installed the latest comitup-lite image on a Pi Zero W. I made no configuration changes. I deleted all related DHCP leases for the Zero's MAC address and rebooted my router.

My iPhone connects to the comitup AP just fine and I get a list of networks to choose from. However, when I enter my PSK, comitup acts as if the PSK was incorrect - it does not connect to the network and I get dumped back at the "choose a network" screen.

When I connect from my Windows 10 machine (Chrome), comitup took the PSK first time and connected to the network perfectly.

```
pi@comitup-694:/var/log $ tail -75 comitup.log
2021-03-17 13:17:31,513 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:17:36,515 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:17:41,527 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:17:46,534 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:17:51,527 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:17:56,542 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:01,546 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:06,553 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:11,556 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:16,562 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:21,557 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:26,568 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:31,577 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:36,582 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:41,586 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:46,592 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:51,591 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:18:56,598 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:19:01,607 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:19:06,608 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:19:11,621 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:19:16,624 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:19:22,530 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:19:27,481 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:19:32,488 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:19:37,516 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:19:42,546 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:19:47,541 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:19:52,563 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:19:57,567 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:03,493 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:08,502 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:13,500 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:18,396 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:18,551 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:23,580 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:28,593 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:33,597 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:38,603 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:43,591 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:48,611 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:53,619 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:20:58,612 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:03,632 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:08,637 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:13,638 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:18,646 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:23,652 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:28,656 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:33,660 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:38,666 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:43,673 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:48,671 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:53,671 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:21:58,686 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:22:03,692 - comitup - INFO - AP active - skipping CONNECTING scan
2021-03-17 13:22:08,828 - comitup - INFO - No candidates - skipping CONNECTING scan
2021-03-17 13:22:13,823 - comitup - INFO - No candidates - skipping CONNECTING scan
[... cut ...]
2021-03-17 13:22:37,692 - comitup_web - INFO - index.html - 14 points
2021-03-17 13:22:37,882 - comitup_web - INFO - index.html - 14 points
2021-03-17 13:22:38,225 - comitup_web - INFO - index.html - 14 points
2021-03-17 13:23:10,751 - comitup_web - INFO - index.html - 8 points
2021-03-17 13:23:12,668 - comitup_web - INFO - confirm.html - ssid pondweed, mode single
2021-03-17 13:23:21,756 - comitup_web - INFO - connect.html - ssid pondweed
```

What else can I provide to assist in troubleshooting this?

davesteele, 2021-03-23:
IOS accesses the web server in a different fashion than other OSs. I just don't use it enough to be able to push this forward much.

jerbzz, 2021-03-23:
Ah, I see. That’s a shame. I’ll add it to the list of “things I would like to submit a PR for in the future”. :)

davesteele, 2021-03-26:
No. I have limited access to iPhones.

davesteele, 2021-04-06:
If your in a position to be able to do this, try the patch on a Zero.

davesteele, 2021-04-06:
Note the new dependency on python3-cachetools.

jerbzz, 2021-04-06:
Awesome - I'll give this a go in a little while and get back to you. Thanks!

davesteele, 2021-04-06:
Just tried it once on a Zero W, with success.

jerbzz, 2021-04-07:
Hi @davesteele 

I think I'm missing something - maybe I don't understand how to properly apply this patch. Here are my steps

- Boot comitup-lite image on Pi 3B, connect to hotspot and then choose network: It works
- `comitup-cli`
- `d`
- Boot same card on Pi Zero W, connect to hotspot and then choose network: It doesn't connect
- Reboot in Pi 3B and go through comitup bootstrap process again
- `sudo apt update`
- `sudo apt install python3-pip`
- `pip3 install cachetools`
- `wget https://github.com/davesteele/comitup/commit/ab4537a7af22b559a040e89094bd4431f5b9d444.patch`
- `cd /usr/share/comitup/web`
- `patch comitupweb.py < ~/ab4537a7af22b559a040e89094bd4431f5b9d444.patch`
- (The first hunk `from cachetools import cached, TTLCache` failed so I had to import it manually)
- `comitup-cli`
- `d`
- Pull the plug to reboot
- Still on Pi 3B here
- Reconnect to hotspot
- I don't get redirected to the captive portal... clearly I broke it!

Am I missing part of the commit?

davesteele, 2021-04-07:
You can "apt-get install python3-cachetools" instead of pip, but that shouldn't matter.
The patch is expecting cwd to be /usr/share/comitup. Not sure why it applied.
For things like this, I tend to check out the repository, and replace the /usr/share/comitup/web directory tree with the repo contents.

jerbzz, 2021-04-07:
OK - got myself sorted out as described.

Now works perfectly on my 3B+ and my Zero W, connecting from Safari on my iPhone 8 Plus (iOS 14.4.2) and from Chrome on my Windows 10 laptop.

Thank you 👍 

I can test other browser and platform combinations if that would be helpful? Ubuntu, Windows, Android...?

davesteele, 2021-04-07:
There's a race. It is just a Performance difference.

Any testing help is appreciated.

jerbzz, 2021-04-08:
Tested the Pi Zero W:
- LineageOS 14.1 with the default browser - perfect
- Firefox 87.0 on Windows 10 - perfect
- Chromium 88.0 on Raspberry Pi OS - didn't get redirected to captive portal but browsed to http://10.41.0.1 and all connected fine :)

davesteele, 2021-04-08:
not  bad. Thanks. Closing

commit 8946c71e (2019-11-10): Fix web connect error from IOS query

When connecting from IOS, the comitup client connect() call fails,
due to another pending D-Bus method on the same connection. Work around
the problem by forcing the connect method to be a separate D-Bus
connection.

Closes #71
https://github.com/davesteele/comitup/issues/71
closed 2021-04-08 by davesteele