davesteele/comitup #75: use non-hotspot for upstream when two wifi interfaces available

CallumJHays, 2019-11-21:
I am currently using a workaround in my `dhcpcd.conf` to accomplish this, but I'm wondering if it should instead be a part of this service?

My `/etc/dhcpcd.conf`:
```
# hack to prevent the hotspot network started by comitup from being the default network when connected to a network with limited access (IAS networks)
interface wlan1
metric 300
```

davesteele, 2019-11-21:
Hotspot dhcp is provided by dnsmasq, managed in [cdns.py](https://github.com/davesteele/comitup/blob/e5b402af592952a88d6813190f0b75e060d4f0f6/comitup/cdns.py#L54). Its scope is limited to the active hotspot wifi interface. Are you saying you have dpcp contention on the upstream connection?

CallumJHays, 2019-11-29:
I'm a little bit confused about all this dhcp stuff - I'm into front-end web and hardware so networking is a bit foreign to me. All I know is that when I'm trying to wget a site hosted on a local wpa2-enterprise network the request doesn't go through. I tried packet sniffing the process with wireshark on the pi and discovered that it was only sending the request over the hotspot interface. I don't have the Pi on me right now but running:
```
routes -n
```

Showed something like:
```
wlan0: metric 301
wlan1: metric 303
```

Which seemed to me to be the cause of the lack of connectivity. Note that at that time and when i was packet sniffing, the wlan1 interface was connected to a sandbox network, as an additional browser-based login was required to access the WAN.

davesteele, 2019-11-29:
You're likely right, it is a routing issue caused by the metrics. 

It would be difficult to debug remotely. About all I can offer is to review the [Installing](https://github.com/davesteele/comitup/wiki/Installing-Comitup) page, or, preferred, start from a [Comitup image](https://davesteele.github.io/comitup/).

closed 2020-06-17 by davesteele