davesteele/comitup #76: Can't connect to wifi using other devices

mrsstaceyseigle, 2019-11-26:
When using iPhone or a Windows computer once I connect to the network named comitup it opens a web page which allows me to select my network. When I select my network it ask for password. Once I enter the password it try to connect however I am redirect back to the page where I have to select the network. (Like it failed to connect)  I make sure I select the correct network and entered the correct password but it just will not connect. 

If I connect the raspberry pi to a monitor then open browser on the raspberry pi and enter 10.41.0.1 it show my network connections. When I select my network and enter password it will connect without a problem. 

Basically it works if you do everything on the Pi however it you use other computers or devices connecting to the comitup network it will not connect

davesteele, 2019-11-29:
See #71 for the iPhone issue. The potential fix is unreleased.

davesteele, 2019-12-01:
Possible/Likely fix is in v1.8 (8946c71)

closed 2019-12-01 by davesteele