davesteele/comitup #81: Start comitup from Python script

Muzzamil1, 2019-11-30:
Hi @davesteele
Is it possible to start the comitup(Access point) after connecting to Wi-Fi using python script with or without deleting the previous configuration?

davesteele, 2019-12-01:
Comitup will delete and re-create the hotspot connection every time it uses it. It does not do that with the upstream configuration.

If you have two wifi interfaces, it is possible to start up the hotspot connection in parallel, via NetworkManager, using "nmcli".

If you start Comitup after upstream is connected, it will (currently) (maybe) tear down the connection for a short time before setting it up again.

Muzzamil1, 2019-12-01:
Ok. So how can I start the Comitup from Python script if pi is already connected to Wi-Fi?
i.e. If I run this in terminal

>  $ `comitup-cli`

 and 

> press d

 it will delete the configuration and start the Comitup. Can I do it from python script? I mean is there any method that I can call from python

davesteele, 2019-12-01:
All of the connections used by Comitup are defined by NetworkManager, meaning that they can be manipulated from Python using the "nmcli" command, or by using NetworkManager D-Bus commands. nmcli is easier. to start with

If you are looking for the equivalent of the "d" command in comitup-cli, you can use (e.g., IIRC) "nmcli con down <connection>" and "nmcli con del <connection>" (not sure if the "down" is needed). If the Comitup service is running ("systemctl status comitup"), it should detect the break and go back to the HOTSPOT state.

closed 2019-12-01 by Muzzamil1