davesteele/comitup #83: Android doesn't obtain IP from hotspot

schweini, 2019-12-01:
I can connect to the AP from my windows laptop, but Android seems to be stuck at waiting for an IP address (Pixel XL, Android 10)

davesteele, 2019-12-01:
I can't replicate from Android 9. The initial experience is usually better if you start from a Comitup image. 

Normally, I'd blame DHCP port contention for the problem, but, in that case, Windows wouldn't work either. If you find the cause of the problem, Let me know.

schweini, 2019-12-01:
Any tips which log files to keep an eye on to identify the problem?

davesteele, 2019-12-02:
Most activity is in syslog.

davesteele, 2020-06-09:
This is likely DHCP server contention. Not a problem in the disk image, and addressed in the package install directions.

closed 2020-06-09 by davesteele