davesteele/comitup #84: Obtaining IP address

Anjum9694, 2019-12-05:
Hi, Having an issue with obtaining IP address from the comitup AP DHCP server. What files should I check?

davesteele, 2019-12-05:
Start with https://github.com/davesteele/comitup/wiki/Installing-Comitup.

Check syslog for DHCP errors.

Did you start with a [Comitup image](https://davesteele.github.io/comitup/)?

Anjum9694, 2019-12-05:
Yes I am running the image it was working fine for a while I installed some programs and suddenly the DHCP server isn't working. The AP shows up fine on android it just shows 'Obtaining IP address'.

Anjum9694, 2019-12-05:
Here is the syslog

Dec  5 13:35:29 raspberrypi NetworkManager[444]: <info>  [1575552929.4236] bluez5: NAP: added interface B8:27:EB:1A:D4:82
Dec  5 13:35:29 raspberrypi bluetoothd[614]: Failed to set privacy: Rejected (0x0b)
Dec  5 13:35:29 raspberrypi nm-dispatcher: req:4 'connectivity-change': start running ordered scripts...
Dec  5 13:35:29 raspberrypi systemd[1]: Started Comitup Web Service.
Dec  5 13:35:29 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[652]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[652]: FAILED to start up
Dec  5 13:35:29 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[653]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[653]: FAILED to start up
Dec  5 13:35:29 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[654]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[654]: FAILED to start up
Dec  5 13:35:29 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[655]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:29 raspberrypi dnsmasq[655]: FAILED to start up
Dec  5 13:35:30 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[656]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[656]: FAILED to start up
Dec  5 13:35:30 raspberrypi avahi-daemon[384]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::ba27:ebff:fee5:2b7d.
Dec  5 13:35:30 raspberrypi avahi-daemon[384]: New relevant interface wlan0.IPv6 for mDNS.
Dec  5 13:35:30 raspberrypi avahi-daemon[384]: Registering new address record for fe80::ba27:ebff:fee5:2b7d on wlan0.*.
Dec  5 13:35:30 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[658]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[658]: FAILED to start up
Dec  5 13:35:30 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[659]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[659]: FAILED to start up
Dec  5 13:35:30 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[660]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[660]: FAILED to start up
Dec  5 13:35:30 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[661]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[661]: FAILED to start up
Dec  5 13:35:30 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[662]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:30 raspberrypi dnsmasq[662]: FAILED to start up
Dec  5 13:35:31 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[664]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[664]: FAILED to start up
Dec  5 13:35:31 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[665]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[665]: FAILED to start up
Dec  5 13:35:31 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[666]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[666]: FAILED to start up
Dec  5 13:35:31 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[667]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[667]: FAILED to start up
Dec  5 13:35:31 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[668]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:31 raspberrypi dnsmasq[668]: FAILED to start up
Dec  5 13:35:32 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:32 raspberrypi dnsmasq[670]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:32 raspberrypi dnsmasq[670]: FAILED to start up
[... cut ...]
Dec  5 13:35:32 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:32 raspberrypi dnsmasq[673]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:32 raspberrypi dnsmasq[673]: FAILED to start up
Dec  5 13:35:32 raspberrypi comitup[487]: dnsmasq: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:32 raspberrypi dnsmasq[674]: failed to create listening socket for port 53: Address already in use
Dec  5 13:35:32 raspberrypi dnsmasq[674]: FAILED to start up

davesteele, 2019-12-05:
Did anything you installed add a DNS or DHCP service, or messed with the configuration of dnsmasq?

In HOTSPOT mode, Comitup will run something like:

`dnsmasq --conf-file=/usr/share/comitup/dns/dns-hotspot.conf --interface=wlan0`

Find out what is taking port 53 on the wifi interface, and why it is running now.

`netstat -elnpu | grep :53`

Anjum9694, 2019-12-06:
Solved by removing recently installed bettercap, it was conflicting with comitup DHCP settings. Just want to ask one more question before closing, where does comitup store current connection config? I mean the settings which comitup store the current selected AP..

davesteele, 2019-12-06:
It doesn't. NetworkManager takes care of that.

muelli, 2019-12-06:
Could comitup watch the dnsmasq process and see it dying? Then it could log an error to the syslog and we'd all know earlier what the problem is.

closed 2020-02-18 by Anjum9694