davesteele/comitup #85: how to disconnect from network and go back to hotspot mode

mrsstaceyseigle, 2019-12-12:
Is there a way to disconnect from a network and go back to hostspot mode.

davesteele, 2019-12-12:
There is a delete command in comitup-cli, or delete the connection with nmcli.

mrsstaceyseigle, 2020-02-05:
comitup-cli then d option gives the option to delete the connections also we can delete the connection with nmcli

Problems. When using nmcli it takes comitup a while to detect and go back to hotspot mode. 

could you please give us the option to input options with the command like
sudo comitup-cli d   or sudo comitup-cli -d

Im using the raspberry pi with a push button to delete the connection

davesteele, 2020-02-05:
Noted.

Try something like "(echo d; echo q) | comitup-cli".

mrsstaceyseigle, 2020-02-06:
Yes (echo d; echo q) | comitup-cli Works great. 

I love to see a enhancement on your next version to use it like " comitup-cli -option"

closed 2020-06-17 by davesteele