davesteele/comitup #87: Regarding usage of applications that may conflict with comitup

Anjum9694, 2019-12-14:
Hi. I have fully functioning comitup running on my RPi and very happy about it. I would like to install Pi-hole but it seems there might be features that could break comitup. Am I good to go or are there any steps I should do before installing? Not just this application but any with a similar circumstance.

https://docs.pi-hole.net/main/prerequesites/#supported-operating-systems

davesteele, 2019-12-14:
If the hotspot is active, the port associated with it serves DNS using dnsmasq. The configuration file is swapped out on-the-fly by comitup when switching between HOTSPOT and CONNECTED modes.That process would conflict with Pi Hole's desire to serve DNS.

The issue could be addressed with judicious editing of the files in /usr/share/comitup/dnc/* (which will be overwritten with each comitup upgrade.

You may be OK if you run with only a single wifi interface. Once connected to your wifi router, comitup's dnsmasq should be uninvolved. You just need to coordinate the services enough to get to that point.

davesteele, 2019-12-14:
I'd love to see a wiki page on how to make this work, if you get to that point.

closed 2020-06-17 by davesteele

davesteele, 2021-04-30:
Here is post on [Comitup plus Pi-hole](http://localhost:4000/raspberrypi/2021/04/29/comitup-and-pihole/)

Anjum9694, 2021-05-01:
I fixed the #141 issue by making a crontab to reset my swap every week, because it seemed to be getting full. @davesteele please check your link, and thanks in advance. I'll check it out.

Anjum9694, 2021-05-01:
Oops @davesteele I meant your post link to the comitup plus Pi-hole.. It is not correct.

davesteele, 2021-05-01:
Oops here too.