davesteele/comitup #88: Forward from eth0 to wlan0

Anjum9694, 2019-12-15:
I'd like the RPi to have internet access when comitup asks for connection to use as a hotspot. Internet connection is provided through the ethernet cable.

davesteele, 2019-12-15:
You'll need to do three things:

- make sure the ethernet connection has the winning default route
- make an iptables nat POSTROUTING rule on the eth interface to turn on masquerading for traffic from the hotspot
- turn on ip_forwarding in /etc/sysctl.conf

Anjum9694, 2019-12-15:
1) I'm not quite sure what that meant.

2) Used `sudo iptables -A FORWARD -i wlan0 -j ACCEPT`
             `sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE`

3) Uncommented ipv4 forwarding line in sysctl.

No internet connection on the hotspot yet..

davesteele, 2019-12-16:
"ip route" to see the routes.
"ip route add default via x.x.x.x"
"ip route del default via x.x.x.x"

Though you should fix the routes using NetworkManager (using e.g. nmtui) to make the changes persistent.

The ip forwarding change requires a reboot (or sysctl -p)

closed 2020-02-18 by Anjum9694