davesteele/comitup #89: Comitupweb templates install directory

PetervdPerk, 2019-12-17:
Hi I'm the creator of https://github.com/PetervdPerk/meta-comitup which provides a Yocto layer for comitup. 

I've created a recipe for comitup that uses setuptools to make a Yocto package. Setuptools installs comitupweb.py to /usr/lib/python3.7/site-packages/web/comitupweb.py and the templates to /usr/share/comitup/web/template. Which causes that comitupweb.py can't find the templates. 

As a hotfix I'm patching comitup using this patch https://github.com/PetervdPerk/meta-comitup/blob/master/recipes-connectivity/comitup/files/web_path_hack.patch

Unfortunately I'm not that advanced in setuptools to find the root cause of this problem that causes that the templates and comitupweb.py don't end up in the same folder. Any suggestions that might cause this problem?

davesteele, 2019-12-17:
I'm happy to see Comitup showing up as a Yocto feature.

For building, I use the "debian" branch. The [makefile](https://github.com/davesteele/comitup/blob/a9b94385443c0b78ea8dc01367fbdec5c23c90a2/debian/rules#L7) causes everything to be installed to /usr/share/comitup, using two options appended to "setup.py install". Then, [links](https://github.com/davesteele/comitup/blob/a9b94385443c0b78ea8dc01367fbdec5c23c90a2/debian/comitup.links) are created to put executables on the PATH, resulting in [this](https://packages.debian.org/sid/all/comitup/filelist). 

I hadn't considered your issue, and would be happy to apply your patch. Would you like to submit a pull request, for proper credit?

You may also be interested in inspecting the customized pi-gen build environment for building [Comitup-flavored Raspbian images](https://github.com/davesteele/pi-gen). Look in [stage2/04-comitup](https://github.com/davesteele/pi-gen/tree/comitup/stage2/04-comitup).

I don't speak Yocto, but is there an issue with versioning? I don't see where the Comitup version number is used for code retrieval. Do you actually want to reference e.g. https://github.com/davesteele/comitup/archive/master/1.9.tar.gz?

davesteele, 2019-12-17:
Nevermind, you're referencing commits.

PetervdPerk, 2019-12-17:
Indeed for now I'm just referencing to commits, Yocto can also use the archives just like debian. For developing using git is a bit easier. Maybe in the future I can refer to archives.

Thanks for pointing the debian branch (completely missed that). Unfortunely Yocto doesn't allow override --install-lib argument (it's hardcoded to site-packages see https://github.com/openembedded/openembedded-core/blob/master/meta/classes/distutils3.bbclass) therefore merging my patch #90 would be a great soiution

closed 2019-12-20 by PetervdPerk