davesteele/comitup #91: BUG: not working after installing

mrsstaceyseigle, 2020-01-20:
I have fresh copy of Rasbian Buster Lite installed on my Raspberry Pi. I have Python, PIP 3 installed as well. I followed your instructions to install. The install say that it complete however the hotspot does not pop up on none of my machines. 

sudo apt-get install comitup

davesteele, 2020-01-20:
If you started with a "Lite" install, and followed [these instructions](https://github.com/davesteele/comitup/wiki/Installing-Comitup), I would expect it to work. Some things to check:

- You should be using a Pi model with built in Wifi. USB adapters typically don't support AP mode.
- If you have a USB adapter plugged in, try it with it removed.
- Do you have evidence that it it booting? At a minimum, the front light should flicker for a while before settling down. If not, look into seeing if the SD card is formatted correctly.
- On a stock install, the Ethernet should negotiate DHCP. Can you coordinate a hard connection to the Pi? If so:
  - Is the comitup service running? (systemctl status comitup)?
  - Are there any errors or Tracebacks in /var/log/comitup.log?
  - Are there any errors in /var/log/syslog or /var/log/daemon.log related to comitup, NetworkManager, dnsmasq, or dhcp?
  - What happens if you use "sudo comitup-cli" to manage connections?
  - Any insights from inspecting "nmcli con", "ip addr", or "ip route"?

Note that I use a [Comitup spin iso](https://steele.debian.net/comitup/). It is pretty much identical to a Raspbian with comitup installed, and /etc/net/work/interfaces edited to remove the Wifi interface.

mrsstaceyseigle, 2020-01-21:
Thank you. The software is up and running. Looks as I forgot this step
3. Rename or delete /etc/wpa_supplicant/wpa_supplicant.conf.

All is working now. Thank you

closed 2020-01-21 by mrsstaceyseigle