davesteele/comitup #95: Source code is different from git clone

mrsstaceyseigle, 2020-02-03:
The source code from git clone https://github.com/davesteele/comitup.git
is different from the source code for this file  	comitupweb.py

davesteele, 2020-02-03:
Yes, recent releases have some changes to the web interface. If you want to see something more familiar, try "git checkout master/<version-without-dash>" or "git checkout debian/<version-with-dash>".

See a git tutorial for more.

mrsstaceyseigle, 2020-02-03:
Is auto redirect after selecting the hotspot included in the versioin. I think the older version had a feature where it would redirect the webpage to the connections page when the hotspot was selected. 

Also I notice that when I edit the  binding address in the web page from 0.0.0.0 to 10.42.0.1 it made the connection page unreachable.

davesteele, 2020-02-03:
"auto redirect" - yes, newer versions should be unchanged in how routing and masquerading is set up between the hotspot interface and the second usb upstream wifi connection.

With newer versions, the ip address of the hotspot was changed to 10.41.0.1, to resolve a reported conflict.

closed 2020-02-04 by mrsstaceyseigle